mantidproject/mantid #13606: Fix auto-scale checkbox when switching to MDSlice

MattKing06, 2015-09-11:
Follow up from #13595.

When switching from 1D view with auto-scale checked to an MDSlice view, the auto-scale checkbox is unticked when it should preserve the state between views.

picatess, 2015-09-11:
The issue arises due to an `endRender` signal of a pqView which is intercepted by the `renderDone` method of the `MdViewerWidget`. 

It is a bit strange that this should only affect the MultiSliceView.

picatess, 2015-09-11:
The difference occurs here:
https://github.com/Kitware/ParaView/blob/master/ParaViewCore/ServerManager/Core/vtkSMVectorPropertyTemplate.h#L248

In the case of the MultiSliceView the passed in values and the internal values seem to differ. In the other views they are the same. Not clear why, but this is probably again something PV internal. The entire process gets triggered by us setting the RGBPoints in the `colorMapChange` slot

PR #13613 (2015-09-11): Fix the multislice view bug for auto scale
Fixes #13606 

@MattKing06  This is a one line fix. Would you mind to have a look at it, since you know the issue already. Thanks.
closed 2015-09-11 by MattKing06