mantidproject/mantid #13620: ResNorm errors with _sqw files as vanadium input

ElliotAOram, 2015-09-14:
Currently the Vanadium file input for ResNorm accepts both _red and _sqw files. When using an _sqw file as the vanadium run and an _res file as the resolution, this produces an error. It may be the case that _sqw should not be used and if so then they should be removed from the valid options for input files in ResNorm. Otherwise, the ResNorm algorithm needs to be updated to accommodate for using _sqw files.

I am currently seeking scientific clarification with Spencer on the issue.

SpencerHowells, 2015-09-14:
As an extra check the workspace/file name should also be tested for a _red ending

closed 2015-09-15 by KarlPalmen

ElliotAOram, 2015-10-05:
This issue has been changed to 3.5 from 3.6 as it was not originally expected to be ready for the release but was.

PR #13630 (2015-09-15): ResNorm error with sqw files
Fixes #13620 

ResNorm should no longer accept non `_red` files as input for vanadium and non `_res` files resolution.

# To Test 
* Open ResNorm (Interfaces > Indirect > Bayes > ResNorm)
* Attempt to add a file for either the Vanadium or Resolution run
* Ensure that:
   * only `_red` files are valid for Vanadium runs
   * only `_res` files are valid for Resolution runs
       * These files should match e.g. `irs26173_graphite002_red.nxs` and `irs26173_graphite002_res.nxs`
* Run the ResNorm Algorithm (click run in the interface) 
* Ensure that the files validate wh [... cut]