mantidproject/mantid #13644: Move MuonAnalysis documentation to mantid help system

raquelalvarezbanos, 2015-09-16:
Move the content of the wiki pages: http://www.mantidproject.org/MuonAnalysis to a new .rst file.

tom-perkins, 2016-03-02:
As part of this, the link on http://www.mantidproject.org/Muon pointing to the MuonAnalysis wiki pages will need to be changed.

tom-perkins, 2016-05-20:
#11731 can be done as part of this issue

tom-perkins, 2016-07-21:
Increasing priority because the documentation will need to be updated when the changes from 
#15518 are merged in. Those updates can be combined with converting the documentation to rst format.

raquelalvarezbanos, 2016-08-03:
@tom-perkins: I've created an issue https://github.com/mantidproject/mantid/issues/17119 to update the links once PR https://github.com/mantidproject/mantid/pull/17118 is merged and we have a new nightly build.

PR #17118 (2016-08-23): Move MuonAnalysis documentation to .rst file
Sulaimaan (work experience student at ISIS) created an .rst file for the `MuonAnalysis` interface.

**To test:**

Build and read the documentation for the `MuonAnalysis` interface. It should look similar to the [wiki page](http://www.mantidproject.org/MuonAnalysisSettings).

I've noticed that in the `Home` tab, section `Detector Grouping`, the option to specify a value for `alpha` does not seem to exist any more, so I've removed the section explaining how to use it. This is the only difference with the wiki page.

Fixes #13644.
Fixes #11731

[Releas [... cut]
closed 2016-08-23 by FedeMPouzols