mantidproject/mantid #13645: Handling errors produced by invalid input for ISIS SANS Interface

MattKing06, 2015-09-16:
**Run Numbers Tab**
- [ ] Error produced when trying to save file from 2D reduction to CanSAS format

**Reduction Settings Tab**
- [ ] Error produced when `Extra Length` field has negative input
- [ ] Error produced when `Radius` field has negative input
- [ ] Error produced when `Wavelength` field has negative input for Logarithmic
- [ ] Verify that input is of numerical type for all necessary input

**Geometry Tab**
- [ ] Mantid Crashes when negative value is used for `Thickness` field
- [ ] Mantid Crashes when negative value is used for `Height` field
- [ ] Mantid Crashes when negative value is used for `Z-offset` field

picatess, 2015-09-16:
Started work on this. How can we hedge against bad values from the User file ?

* Geometry Tab*
* Patching line edit fields with validators
* z-offset should probably allow negative values --> need to investigate further here

Mantid-Matthew, 2019-08-16:
This issue related to the parsing in the old GUI

closed 2019-08-16 by Mantid-Matthew