mantidproject/mantid #13663: ISIS Reflectometry GUI options need documentation update

MichaelWedel, 2015-09-16:
This was discovered when following the testing instructions for unscripted testing for the Reflectometry GUI on Ubuntu 14.04.

Compared to the documentation on the Wiki, the "Options" dialog of the ISIS Reflectometry interface has different options now, the documentation should be updated regarding what these options mean. Screenshot of the dialog as found in the unscripted testing:

[image]

Also, the first option was now checked by default, while the testing guide mentions it should not be checked.

OwenArnold, 2015-10-12:
Updated documentation [here](http://www.mantidproject.org/ISIS_Reflectometry_GUI#Options_Dialog)

MichaelWedel, 2015-10-14:
Updated documentation explains new options. Screenshot is still the original one, but since you mentioned that this interface is probably going to be deprecated it's not urgent/necessary to replace it.

closed 2015-10-14 by MichaelWedel