mantidproject/mantid #13665: Muon ALC Interface incorrect display of system busy icon on showing user error

KarlPalmen, 2015-09-16:
When ALC is opened and I select invalid data files in ALC analysis and the error message is shown, the system busy icon (was hourglass in older version of Windows) is shown although it should not be shown then. This may occur on other invalid user input.

raquelalvarezbanos, 2015-10-08:
I've been thinking about this and come to the conclusion that we should remove the system busy icon. The reason is that this icon suggests that the user should wait while data are being loaded. However, as soon as #13895 is merged, this will no longer be the case. We should however disable the input properties (dead times, grouping, etc), so the user can't change them while loading data. I am going to create a new issue and close this one.

KarlPalmen, 2015-10-08:
Superseded by #13899 .

closed 2015-10-08 by KarlPalmen