mantidproject/mantid #13699: Update ipython and matplotlib to make nbagg backend available in ipython notebooks

matthew-d-jones, 2015-09-18:
The nbagg matplotlib backend allows interactive plots in ipython notebooks. Unfortunately, we do not have the required versions of matplotlib and ipython in the mantid python environment on any platform.
I have marked the issue as high priority as this is a requested feature, and important to gaining feedback on support for ipython notebooks in Mantid, see #10085.

Requirements:
matplotlib version >=1.4.3
ipython version undocumented, but 3.2.1 is known to work
jsonschema

martyngigg, 2016-10-14:
Windows is fine here now and Ubuntu 16.04 has all of the required versions for matplotlib and the notebook.

I propose that we also don't do anything for RHEL6  package wise and if users want it there we provide `pip` install instructions.

That leaves RHEL 7 & OSX. RHEL 7 has IPython 3.2.1 and we should get `matplotlib` 1.5.1 in #16873 but it does not have the notebook as it's version of `tornado` is too old. @stuartcampbell What do you think we should do here?

@quantumsteve Any suggestions for OS X?

quantumsteve, 2016-10-14:
@martyngigg We could use the formula in [`homebrew-python`](https://github.com/homebrew/homebrew-python) and copy its contents into the .app bundle. 

The other challenge we'll have is ensuring that our matplotlib is before the system library in the `pythonpath`.

quantumsteve, 2016-10-14:
matplotlib 1.5 requires Python 2.7. Do we require that on RHEL 6?

martyngigg, 2016-10-26:
@quantumsteve I was able to `pip` install matplotlib 1.5 on RHEL 6 with python 2.6 and it seemed to work.

The option of using the `homebrew-python` formula seems to make sense. Is it enough to update https://github.com/mantidproject/mantid/blob/master/MantidPlot/src/PythonScripting.cpp#L219 for the `sys.path` handling?

stuartcampbell, 2016-11-22:
@martyngigg I have working `python-tornado` and `ipython-3.2.1` (with notebook) packages for RHEL7.  Should I copy them into the mantid copr repo ?

stuartcampbell, 2016-11-22:
Also, just for completeness.  The newer version of `python-tornado` is going to be included in RHEL 7.4 so the Fedora/EPEL version of ipython can then ship with the notebook.

martyngigg, 2016-11-22:
@stuartcampbell That sounds good.

martyngigg, 2017-10-31:
I'm marking this as resolved as the packages have been updated in all places that require them. We have training material based on the notebooks: https://github.com/mantidproject/training and they are in active use by the reflectometers at ISIS.

closed 2017-10-31 by martyngigg