mantidproject/mantid #13736: Script Repository GUI sees incorrect status of file

martyngigg, 2015-09-23:
James Lord at ISIS reported a bug with the script repository:

"I’ve made the change to CreatePhaseQuadTable described below and was about to upload the revised version to the Script Repository, where the earlier one had already been stored since January this year. However the Script Repository dialog seemed to think that the one on the server was newer and offered to download it again (orange down arrow).
 
(Is this a bug in the beta 3.5 version of Mantid I’m testing – 3.4.20150915.1737 - or a general feature of the Repository server?)
 
I let it download the old one again (overwriting the saved new one) and then overwrote it by re-saving the new one from the Python Window, then it let me upload it (blue arrow). I think all is OK now (green tick).
 
As an aside should the Script Repository have two local directories (or trees) – one for those I’ve downloaded and find useful, and another for ones I’ve developed myself and will upload? Then there’s no confusion.
 
James."

I know there was a fix for the status of the uploaded files so we need to check if we have broken something.

matthew-d-jones, 2015-09-23:
I have been able to reproduce the bug with the following steps (using the test script repo):
- Start MantidPlot
- Download a file from the repository
- Close MantidPlot
- Delete the repository json files
- Edit the file
- Start MantidPlot, create and view script repository

martyngigg, 2015-09-23:
Excellent. Are you able to look into fixing it?

matthew-d-jones, 2015-09-23:
Yes looking into it now. I just wanted to record how to reproduce it, for myself and for the person who tests the pull request when I've fixed it.

matthew-d-jones, 2015-09-23:
After some consideration I think this is actually not a bug, but we do need to indicate to users what is going on... it took me a considerable amount of time to figure it out from the code.

I think that the situation is that the user has created a new repository, probably in the process of installing beta 3.5. They then transferred a modified script to the repository, without first downloading it from the server. Because we have no record of the file being downloaded, it is impossible for us to know if the script was modified only locally or both locally and remotely. The behaviour is therefore to display the status as "changed remotely" and, if the user opts to download the script, a backup version of the local file is created as well.

NickDraper, 2015-09-30:
Mattew can you make sure to feed back to James Lord what the cause of the problem seems to have been.  Your tooltip change can go into R3.6

matthew-d-jones, 2015-10-01:
I've emailed James Lord with an explanation of what I believe the cause to have been, but have asked him to contact me if he comes across the problem again.

PR #13832 (2015-10-05): Re #13736 Improve tooltip on status for local and remote change
Closes #13736 

Clearer tooltip on the status button to distinguish between the cases of a file having changed remotely only, or both remotely and locally. In the latter case the remote file is offered for download, but the tooltip informs the user that a backup of the local version will be created.

Tester:
Code review may be sufficient, but if you want to see the tooltips in context then you can use the following steps:

- Download any file from the script repository
- In the .local.json file in your script repository directory, [... cut]
closed 2015-10-05 by picatess