mantidproject/mantid #13738: Fix run information in MuonAnalysis interface

raquelalvarezbanos, 2015-09-23:
The MuonAnalysis interface allows co-adding different runs to improve statistics: multiple runs/files can be specified, in which case they will all be summed to form a single data set. However, some of the information displayed in the 'Run Information' corresponds to the first run only.

This was reported by James Lord:

> In the Run Information box at the bottom of the dialog (not shown in your screen shot), the total counts and frames are correct but the start/end times and average temperature etc are for the first run only. The “Runs” line assumes a consecutive sequence of runs even if I select a few nearby ones with gaps. The “Sample Temperature” and “Sample Magnetic Field” header values ought to indicate whether the runs are at the same nominal value, and if not then show a range e.g. “Sample Magnetic Field: 1000.0 to 1020.0”. The values derived from the logs (Average Temperature) can simply concatenate the logs and then average them. Start Time and End Time can be the earliest start and latest end.

Another bug in the 'Results table' was reported by James:

> Having fitted a function to the summed data, in the “Results Table” tab the range of runs is shown, but if I create a table it’s abbreviated back to the first run in the set. Is this because the table is numeric, despite the run number column being type “Label”?

tom-perkins, 2015-12-23:
Break issue down into tasks:
- [ ] Runs label should deal with non-consecutive ranges
- [ ] Sample temperature, magnetic field should be a range if not all at the same value
- [ ] Average temperature should be from all logs
- [ ] Start, end times should be earliest start and latest end
- [ ] Results table should show range of runs, not just the first

tom-perkins, 2016-03-03:
The "result table" point above links to point 5 in #13096: 
> If fitting more than one period or combination from the same run, there's only one entry in the Results table. Should all of them be there, perhaps with an extra column for period number?

At the muon meeting 25/2/16 it was decided to use the format `run number: period` for this, and leave off the period for All or single-period data.

closed 2016-03-30 by FedeMPouzols