mantidproject/mantid #13784: Engg algorithms and GUI: more additions - modes for multi-run focusing

FedeMPouzols, 2015-09-28:
Following on other required improvements to the Engg GUI: #13376 (most of which if not all should be implemented first), and other improvements listed in #13783. Add these (which were left out of #13486):

- Combo box for the "multi-run focus mode" when you specify multiple runs. Two alternatives: `focus individual run files separately`, `focus sum of files`. This would improve with the next point:
- Context help to explain the advanced use of the multi-ranges of runs.

Note this requires the MWRun widgets to be added (#13376).

shahrozahmed1, 2015-10-15:
Note: go first for the sequential multi-run processing, then the "sum of files" option is a lower priority.
For the "sum of files", consider algorithms like for example `Plus` and `MergeRuns`.

FedeMPouzols, 2015-10-15:
Note: this will require changes in `EnggDiffractionPresenter::plotFocusedWorkspace` which now assumes that you are just plotting banks.

FedeMPouzols, 2015-11-03:
This is not closed yet...

PR #14186 (2015-11-03): Engg:: miscellaneous additions and improvements
 Fixes #13784

> * [x] merge #14167 before this

*Set-up for the GUI can be found here #14167*

Saves to new default path when calibration files are generated and creates the path if not found;
`C:\EnginX_Mantid\User\<RBNumber>\Calibration` 
(Where all the calibrations files are saved)
PR #14579 (2015-11-26): Engg::Enabled Multi-Run Focusing for Focus Tab with Two Different Focusing Modes
fixes #13784 

- Enables the focusing tab to take multiple files and also option of utilising two different type of focusing
- Presenter test has also been updated

*Note That*
- The `Plot Data Representation` is not compatible with the multi-run focusing yet
- `Focus Sum Of Files` will not be functioning on this branch

**To Test**
`Interface/Diffraction/Engineering Diffraction`
*Get initial settings from [here](https://github.com/mantidproject/mantid/pull/13501)*

Provide any of three focusing text field with a  [... cut]
closed 2015-11-26 by FedeMPouzols