mantidproject/mantid #13795: SANS GUI: Implement sasCIF output format

smk78, 2015-09-29:
The biological SAS community have long been disadvantaged (compared to the MX community) because the Protein Data Bank does not routinely accept the deposition of data and structures derived from solution scattering. This has led to the creation of other databases but these have been hampered by issues surrounding acceptable deposition file formats. In 2012 the SAS Commission of the IUCr set up a taskforce to provide some leadership and make a recommendation. The taskforce reported back at SAS-2015 in Berlin earlier this month. The recommendation is that the biological SAS community use the sasCIF format.

sasCIF was first developed at the EMBL around 1999-2000, and was actually implemented in the ISIS COLETTE data reduction program. It has been through a number iterations since and now has the backing of the CIF Commission (see http://www.iucr.org/resources/cif). Consequently the time is probably right for sasCIF to be added to the available output formats in Mantid.

To facilitate this, a Python toolset/library is now available - https://github.com/mkachala/sasciftools - although this requires users to have the EMBL ATSAS suite installed.

Implementing sasCIF will (based on earlier experience) likely be tedious, but it woul dmake a nice self-contained project for someone.

gemmaguest, 2019-10-02:
@DavidFair Ask Steve if this still needs doing

smk78, 2019-10-03:
It is not a priority issue (from my perspective). But I don't think we should lose sight of it just yet.
I guess what probably needs to happen is for someone to contact the SANS scientists at other Mantid partner institutions and see how big a clamour there is? Sounds like a job for me, doesn't it?! :-)

closed 2021-03-01 by stale[bot]