mantidproject/mantid #13801: SANS GUI: Error when using .n001, .n002, etc files as transmissions

smk78, 2015-09-29:
“update and store” files, with extension .n001, .n002 etc  will reduce when chosen as scattering (SANS), but do not work as transmission files where they give a “spectrum number” error.

Suspect Mantid is looking at the wrong workspace. Note the compromise workspace names 1_sans_001 and 1_sans_001_monitors. Can supply examples if needed.

This means that we cannot reduce on the fly the current data collection when we are using M4 beam stop transmissions (unless we can find some completed run with similar transmission for a quick look).

Richard

gemmaguest, 2019-10-02:
Ask Steve if we still want to do this

smk78, 2019-10-03:
As this is quite an old ticket, I've just tried this in Release 4.1.0 on some LOQ 19/1 data: SANS/TRANS 106895.n001 and Direct 106892. You can process the .n001 as just a Sample SANS, but if you add it in as  the Sample TRANS you get this:

```
Unable to successfully fit the data: Function doesn't depend on parameter A0
Error in execution of algorithm CalculateTransmission:
Unable to successfully fit the data
Error in execution of algorithm SANSCalculateTransmission:
Unable to successfully fit the data
  at line 99 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCalculateTransmission.py'
  caused by line 165 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCalculateTransmission.py'
Error in execution of algorithm SANSCreateAdjustmentWorkspaces:
Unable to successfully fit the data
  at line 99 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCalculateTransmission.py'
  caused by line 165 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCalculateTransmission.py'
  at line 116 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCreateAdjustmentWorkspaces.py'
  caused by line 207 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCreateAdjustmentWorkspaces.py'
Error in execution of algorithm SANSReductionCore:
Unable to successfully fit the data
  at line 99 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCalculateTransmission.py'
  caused by line 165 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCalculateTransmission.py'
  at line 116 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCreateAdjustmentWorkspaces.py'
  caused by line 207 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCreateAdjustmentWorkspaces.py'
  at line 119 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSReductionCore.py'
  caused by line 187 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSReductionCoreBase.py'
Error in execution of algorithm SANSSingleReduction:
Unable to successfully fit the data
  at line 99 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCalculateTransmission.py'
  caused by line 165 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCalculateTransmission.py'
  at line 116 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCreateAdjustmentWorkspaces.py'
  caused by line 207 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSCreateAdjustmentWorkspaces.py'
  at line 119 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSReductionCore.py'
  caused by line 187 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSReductionCoreBase.py'
  at line 110 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSSingleReduction.py'
  caused by line 126 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSSingleReductionBase.py'
  caused by line 150 in 'C:/MantidInstall-4.1.0/plugins/python/algorithms\WorkflowAlgorithms\SANS\SANSSingleReduction.py'
  caused by line 154 in 'C:\MantidInstall-4.1.0\scripts\SANS\sans\algorithm_detail\single_execution.py'
'''

which looks to me like it's not calculating a transmission so the transmission fitting falls over?

This needs fixing.

smk78, 2020-07-16:
Any cross-over with https://github.com/mantidproject/mantid/issues/24407?

smk78, 2020-07-16:
Also see https://github.com/mantidproject/mantid/issues/27482

smk78, 2021-04-21:
I've just retried this issue in Mantid 6.0 as follows:

Using data in \\isis\inst$\NDXLOQ\Instrument\data\cycle_19_1 and User file \\isis\inst$\NDXLOQ\User\masks\USER_LOQ_191C_M4_Simm_6mm_Changer_MERGED_2p2-10Ang:

Sample_scatter 106895.n001, Sample_transmission 106895.n001 Sample_direct 106892  >>>  works fine

Using User file \\isis\inst$\NDXLOQ\User\masks\USER_LOQ_191C_M3_Simm_6mm_Changer_MAIN_2p2-10Ang and:

Sample_scatter 106895 Sample_transmission 106892 Sample_direct 106979    >>>  works fine

Making a copy of transmission file 106892 and renaming it as .n001:

Sample_scatter 106895 Sample_transmission 106892.n001 Sample_direct 106979    >>>  works fine

So for **histogram** data this issue would seem to be fixed.

smk78, 2021-04-21:
So then tried ZOOM data from \isis\inst$\NDXZOOM\Instrument\data\cycle_20_3 and User file \isis\inst$\NDXZOOM\User\masks\USER_ZOOM_Nylander_4m_SampleChanger_203E_12mm_Large_BEAMSTOP which uses M4 for transmissions:

Sample_scatter 17752, sample_transmission 17846, sample_direct 17843  >>>  works fine (as you’d expect)

Make a copy of 17846.nxs as 17846.n001 and do:

Sample_scatter 17752, sample_transmission 17846.n001, sample_direct 17843  >>>  works fine

Then switch to \isis\inst$\NDXZOOM\User\masks\USER_ZOOM_Nylander_4m_SampleChanger_203E_12mm_Large_BEAMSTOP_M5 to use M5 transmissions:

Sample_scatter 17752, sample_transmission 17752, sample_direct 17859  >>>  works fine (as you’d expect)

Make a copy of 17752.nxs as 17752.n001 and do:

Sample_scatter 17752, sample_transmission 17752.n001, sample_direct 17859  >>>  works fine

_EXCEPT_ the M5 calculation is not giving the same result as the M4 calculation. I don't know why! The M4 is correct.

[image]

But from the perspective of this particular issue, using .n001, etc files as transmission files would seem to be working in 6.0.

gemmaguest, 2021-05-04:
Closing because this is working in v6.0. We won't worry about the M4/M5 calculation difference unless this is reported by the Zoom scientists - Steve thinks it's possible he was just using a weird file.

closed 2021-05-04 by gemmaguest

smk78, 2023-08-21:
Have retested this with histogram data in MantidWorkbenchNightly-6.7.20230816.1558. Still appears to work. Following data package can be used to verify.
[n001-test.zip](https://github.com/mantidproject/mantid/files/12396061/n001-test.zip)

I have noticed a small issue with the metadata reporting which I'll add to https://github.com/mantidproject/mantid/issues/28998.