mantidproject/mantid #13802: reference to ‘uint64_t’ is ambiguous in LoadFITS.cpp

leomrtns, 2015-09-29:
Hello, while compiling mantidnightly-3.4.20150918.2040, I found the following error: 
```
/usr/local_src/mantidnightly-3.4.20150918.2040-Source/Framework/DataHandling/src/LoadFITS.cpp:278:8: error: reference to ‘int64_t’ is ambiguous
```

Followed by a complaint about candidates from both include/stdint.h and include/boost/cstdint.hpp . I got the same error for uint64_t.

The issue was easily fixed by prefixing the types with "std::", that is, `std::uint64_t` and `std::int64_t` in this file.

FedeMPouzols, 2015-10-01:
Hi @leomrtns . Thanks for the report. This hadn't been caught by our build servers. I presume you were compiling with g++ 4.9.x?

leomrtns, 2015-10-01:
Hi @FedeMPouzols thanks for the prompt action. Yes, I am using g++ 4.9.2. Other than that it compiled smoothly. (I also had to add a few paths by hand to `launch_mantidplot.sh` and change the executable name on it, besides copying `lib/paraview-4.3/` from the stable distribution, but I am sure this is due to some mistake on my part.)

PR #13831 (2015-10-05): Remove too wild using namespace(s) from LoadFITS
Fixes #13802. 

`using namespace std` and `using namespace boost` were producing ambiguities with uint types. This PR removes them both.

*To test*: this should eradicate namespace pollution once for all. Code review should be enough.
closed 2015-10-05 by martyngigg