mantidproject/mantid #13816: Add a few missing (optinal) variables in the files saved by SaveOpenGenieAscii

FedeMPouzols, 2015-09-30:
This does not look very high priority, and it may not be needed. In principle the OpenGenie scripts for ENGINX are working well, but just complaining that these fields/variables are not found:

- xunits
- run_no
- spec_no
- title
- ylabel

Some of which are sometimes included in the logs. 

`spec_no` definitely needs to be provided as an optional (default: empty) input property (string type) to `SaveOpenGenieAscii`. It looks like "1201-1400". 

The others could be generated by `SaveOpenGenieAscii` using information from the workspace (or similarly named logs which are often found, like for example "run_number". But if we add these 5 variables that should be enabled/disabled via an input property. The 5 variables seem to be expected by the EnginX scripts in particular but are not always present as log entries in the run files tested.

shahrozahmed1, 2015-10-02:
@FedeMPouzols Could this possibly be merged to 3.5? as its only small update and could be useful not to have too many errors when you input the file to OpenGenie

FedeMPouzols, 2015-10-02:
@shahrozahmed1 : unless the OG scripts fail without this update I'd be conservative and leave this PR for 3.6. In principle the version that was already merged for 3.5 is already usable, and as there are some last minute issues being fixed right now let's minmize changes and new builds unless absolutely necessary.
The EnginX scientists will be using nightly builds, as there's many updates coming soon. So in the worst case this would just imply a delay of a few days for them.

PR #13834 (2015-10-05): SaveOpenGenie::Add missing variables & included different formatting style
Fixes #13816 

**To Test**

Following variables can be found in ascii file when `ENGIN-X Format` is applied:
- xunits
- run_no
- spec_no
- title
- ylabel

Following variables can be found in ascii file when `Basic Format` is applied:
- run_no
- title

SpecNumberField only accepts valid format which is something like... `1200 - 2000`.
SpecNumberField is **not** accessible when basic format is selected in the `SaveOpenGenieAscii` input dialog.
closed 2015-10-05 by FedeMPouzols