mantidproject/mantid #13824: Improve documentation of outputs from Fit when using multi-domains

FedeMPouzols, 2015-10-01:
This is not detailed in the current RST of `Fit`. How do you get the outputs, especially when you're using it from a script? The indices used for the suffixes of the names of the output workspaces start with _0 or _1?
It also seems that the WorkspaceGroup is only created if the output property is assigned?

This could go below the section `Output`, for example a new section/subsection called `Output when fitting multiple functions`.

FedeMPouzols, 2016-07-13:
A recent change that may be relevant when clarifying how `MultiDomainFunction` works and their capabilities: #16837.

NickDraper, 2017-06-02:
Resolved by recent PR by anthony Lim adding examples

closed 2017-06-02 by NickDraper