mantidproject/mantid #13835: Unexpected exception when deleting items in a workspace group

raquelalvarezbanos, 2015-10-01:
Load a workspace group, for instance, load runs MUSR00015189 to MUSR00015193. A workspace group will be created. Remove one of the items (not the last one) and delete the workspace group, I get the following unexpected exception:

> WorkspaceGroup - index out of range. Requested = 5, current size = 5

You can select "Continue" but Mantid will crash as soon as you try to do something else.

I've noticed this bug when working with workspace groups. This seems to happen only if the group has a sufficiently large number of items

LamarMoore, 2015-11-30:
I am unable to reproduce this issue.

raquelalvarezbanos, 2015-11-30:
Hi @LamarMoore if this is no longer an issue, could you mark it as "Review required"?

closed 2015-12-01 by NickDraper