mantidproject/mantid #13838: Corrections for images from the MCP detector

FedeMPouzols, 2015-10-02:
This is needed for energy selective experiments. Correct the shifts of the sensors/tiles, and the gaps between them. Code examples for how this is done at SNS has been provided by @JeanBilheux.

Requirements (may spawn more issues):
- Having these corrections as an algorithm would be handy.
- Other scripts/algorithms that aggregate energy bands need to use this to generate stacks of corrected images that can be directly passed on to reconstruction scripts.
- We'll also need standalone Python code that implements this for when we run the "many energy bands"=>"single stack" aggregation process on remote machines (SCARF).
- Should work on traditional workspaces and also for the new "Lean" MD workspaces when they become available.

FedeMPouzols, 2016-09-22:
@DimitarTasev: implementing this requires feedback and some experiments from the instrument scientists.

DanNixon, 2017-10-10:
This will be done as part of [`mantidimaging`](https://github.com/mantidproject/mantidimaging).

closed 2018-03-02 by DanNixon