mantidproject/mantid #13844: Investigate crash with OpenCascade 6.9 on OS X.

quantumsteve, 2015-10-02:
Linking against OpenCascade 6.9 causes the following unit tests to fail:

	144 - GeometryTest_InstrumentDefinitionParserTest (Failed)
	441 - PythonAlgorithms_ISISIndirectEnergyTransferTest (SEGFAULT)
	458 - PythonAlgorithms_OSIRISDiffractionReductionTest (SEGFAULT)
	700 - AlgorithmsTest_AppendSpectraTest (Failed)
	728 - AlgorithmsTest_ConjoinWorkspacesTest (Failed)

The crash report from PythonAlgorithms_ISISIndirectEnergyTransferTest is available [here](https://gist.github.com/quantumsteve/8287afdb1a564ccf11c6)

quantumsteve, 2015-10-07:
The issue only appears when OpenCascade is built with optimization enabled. All tests pass after adding `ENV.O0` to the homebrew script for OpenCascade. 

According to [this bug report](http://tracker.dev.opencascade.org/view.php?id=26042) the issue is a new compiler optimization added in XCode 6.3. It has been [fixed in master](http://git.dev.opencascade.org/gitweb/?p=occt.git;a=commitdiff;h=0e14656b3052946192bbb08a998e9c23a3be8752). I have downloaded the snapshot and verified that the SEGFAULT no longer occurs.

Given both the size of this patch and the [substantial changes](http://dev.opencascade.org/index.php?q=node/1091) between OpenCascade 6.9.1 and OpenCascade 7.0 (Scheduled For Release on or around 2015-11-30), backporting this patch is a considerable amount of work with little gain. 

Since this issue doesn't currently affect our build servers (which use XCode 6.2), I think the best course of action is for developers using Yosemite or El Capitan to build OCE or OpenCascade without optimizations and to upgrade to OpenCascade 7.0 when it becomes available.

peterfpeterson, 2015-10-08:
It doesn't feel like a good answer, but it appears to be the best given the options.

closed 2015-10-08 by peterfpeterson