mantidproject/mantid #13909: "Ask for help" bug on RHEL7

AndreiSavici, 2015-10-08:
In MantidPlot, when clicking on Help->"Ask for Help", it is supposed to be opening the webpage of the mantid forum. On RHEL7 machines, it opens for a second, then the browser automatically shuts down. If I already have firefox open, a new tab is created, and everything is fine. I saw the same problem if I try to click on Help->Mantid Homepage.

martyngigg, 2016-10-14:
@AndreiSavici Is this still an issue?

AndreiSavici, 2016-10-14:
It still is

martyngigg, 2016-10-14:
So it looks like this is a problem when trying to start firefox when using tcmalloc with LD_PRELOAD - https://bugzilla.mozilla.org/show_bug.cgi?id=866791

If I use chrome as the default then everything is okay. I'm not sure how to get around this...

martyngigg, 2016-10-14:
I can also reproduce this on Ubuntu if I set Firefox as my default browser and launch MantidPlot through `bin/launch_mantidplot.sh`.

PR #17887 (2016-10-26): Implement wrapper for QDesktopServices
Adds a wrapper class for `QDesktopServices` so that we can intercept the `openUrl` call and remove the `LD_PRELOAD` environment variable when starting the browser on Linux-based systems. Firefox crashes if tcmalloc is preloaded when it starts.

**To test:**

Needs to be tested on Linux. 

* set Firefox as the default browser through the Firefox preferences menu.
* close any running instances of Firefox
* start mantidplot through `bin/launch_mantidplot.sh` and click "Help->Ask for Help" - Firefox should so the help documentation. 

Fixes #13909.

[RELEASE  [... cut]
closed 2016-10-26 by quantumsteve