mantidproject/mantid #13950: Logically dead code in FindPeaks.cpp

NickDraper, 2015-10-13:
FindPeaks.cpp Line 1088 onwards

This was highlighted by coverity 1301341

A change you made in this commit
https://github.com/mantidproject/mantid/commit/d5970611f3c279d64d185ca7e1f2a8b7011fb774

meant that the change to the value of fitresult was hidden from the variable of the same name does not occur.
This means the large if statement lines 1101 - 1140 will never run.

Please verify that was your intention, and correct if it is not.

peterfpeterson, 2015-10-15:
A first attempt at this is in #13667.

FedeMPouzols, 2015-10-16:
I can confirm that if you try to fix this issue by not hiding/redefining the variable then several unit tests break (besides AlgorithmsTest_FindPeaksTest):

- PythonAlgorithms_EnggCalibrateTest
- PythonAlgorithms_EnggFitPeaksTest
- MDAlgorithmsTest_IntegratePeaksMD2Test
- MDAlgorithmsTest_IntegratePeaksMDTest
- PythonAlgorithms_TOSCABankCorrectionTest

And I'd anticipate that several or maybe many system tests will fail as well.

I know the first two tests (Engg ones) and they way the get broken doesn't look good (a totally wrong fit result because of wrong peaks found in one case, and lots of NaN/Infinity problems in the second case).

Refs #13667.

FedeMPouzols, 2016-04-26:
This is also a cppcheck (1.73) issue: #15978.

closed 2019-07-25 by NickDraper