mantidproject/mantid #13988: Add vtkUniformGrid option to the VSI

quantumsteve, 2015-10-14:
Add the option to use a [vtKUniformGrid](http://www.vtk.org/doc/nightly/html/classvtkUniformGrid.html) when the binned axes are orthogonal (angle between them is 90 degrees).

NickDraper, 2016-02-19:
What benefit would this give, and what should we call it "in user speak"

quantumsteve, 2016-02-19:
@NickDraper "in user speak" this should both dramatically reduce memory consumption and rendering time when slicing, contouring and volume rendering large MDWorkspaces. 

`vtkUniformGrid` inherits from `vtkImageData`, and NIH is funding substantial work to improve rendering  this data type (used by, for example, medical imaging). `vtkUniformGrid` is also used in VTK's native AMR datasets, similar to the MDEventWorkspaces in mantid.

OwenArnold, 2016-06-24:
This looks like a nice feature that Paraview are now providing ("Resampling to Image"). Following on from what Nick said, I think we need to discuss how this would work for the end-user. The advantage of the current orthogonal binning is that a rebinned workspace is also generated, while this approach would be be in-visualisation only. We need to make sure that our naive users are not burdened with having to understand the difference between these two options. I think it could be very worth while to add this option, but from feedback we've had from our users in the past, we need to be careful about adding any complications.

quantumsteve, 2016-08-10:
I showed this to @AndreiSavici yesterday. While I don't yet have quantitative numbers, the frame rate before and after "Resampling to Image" is striking. Running on my desktop, it went from a frame every few seconds to being essentially fluid.

The non-orthogonal axes were still present after the filter was applied. Image quality is a bit tricky. The diffuse scattering planes were harder to see on the interpolated cartesian grid.

OwenArnold, 2016-08-11:
Sounds promising. One of the criticisms of the VSI was always that rendering was slow. 

Ideally we want to switch between vtkStructured/vtkUnstructured and vtkUniformGrid according to user inputs/data (but not explicitly stating one or the other), and hide from the users the need to know anything more about outputs.

quantumsteve, 2016-08-19:
https://blog.kitware.com/dataset-resampling-filters/

closed 2021-03-03 by stale[bot]