mantidproject/mantid #13996: ISIS powder diffraction - settings/preferences file

FedeMPouzols, 2015-10-15:
Together with the workflow diagram done in #13408, we'll also need a good understanding of the options included in the `mtd.pref` file and similar `pref` files.

This can be as simple as the list of options that are found in the example/test `mtd.pref` files, but we need to make sure we understand the role of each of them, and in what steps / algorithms they are used.

shahrozahmed1, 2015-10-20:
This issue has now been resolved, the solution can be found [here](https://github.com/mantidproject/documents/tree/master/Design/PowderDiffraction_ISIS)

FedeMPouzols, 2015-10-20:
This sheds a lot of light on how these scripts work. There's:
- A couple of tables of options/variables from the preferences file (`mtd.pref`) added in the README
- PowderDiff_AlgorithmFlow: an algorithm flow diagram that links the Mantid algorithms with the scripts/sequence of calls for HRPD.
- VariablesUsage: detailed table of where in the code the more important preferences/variables are used.

@shahrozahmed1 : can you add a bigger PowderDiff_AlgorithmFlow, a better resolution pdf as with the other documents (or any other format) so it prints nicely on paper?

closed 2015-10-20 by FedeMPouzols

shahrozahmed1, 2015-10-20:
@FedeMPouzols I have uploaded [pdf version](https://github.com/mantidproject/documents/blob/master/Design/PowderDiffraction_ISIS/PowderDiff_AlgorithmFlow_pdf.pdf) of PowderDiff_AlgorithmFlow