mantidproject/mantid #14000: Allow ReflectometryReductionOne's ProcessingInstructions property values to be specified as workspace indices or spectrum numbers

matthew-d-jones, 2015-10-15:
For the ReflectometryReductionOneAuto algorithm there has been some confusion over what the ProcessingInstructions property values are, see #13828. They are currently workspace indices, not spectrum numbers or detector IDs.

The suggested change is to add the ability to prefix the values in the ProcessingInstructions property string with something to indicate whether the given values are workspace indices or spectrum numbers, e.g. ProcessingInstructions="sp=167-201" or ProcessingInstructions="wi=167-201".

It has been requested that the default, when values are supplied with no prefix, corresponds to what is displayed in the data cursor for the spectrum and slice viewers. Probably spectrum numbers (check this). It may be worth only adding the prefix functionality for the next release and informing users that the default will change from workspace indices to spectrum numbers in the following release.

matthew-d-jones, 2015-10-15:
@OwenArnold Anything to add, or that I should clarify?

OwenArnold, 2015-10-15:
@raquelalvarezbanos This might be a good ticket for you to pick up since yo will take over oversight of the Reflectometry Technique are for ISIS shortly.

raquelalvarezbanos, 2015-10-15:
Thanks @OwenArnold I'll let you know when I start working on it.

PranavBahuguna, 2017-07-25:
I'm pretty certain this has been addressed, closing now.

OwenArnold, 2017-07-25:
This would benefit from the use of the workspace index property that @LamarMoore is working on currently.

PranavBahuguna, 2017-07-25:
On closer inspection it seems that there is still work to be done in this regard. Reopening.

PR #23431 (2018-11-06): RRO/RROA Processing instructions switched to taking spectrum numbers instead of workspace indices
~~~THIS SHOULD ONLY BE MERGED AFTER #23341~~~

**Description of work.**
The processing instructions property of the ReflectometryReductionOne2 and ReflectometryReductionOneAuto2 algorithms now take in spectrum numbers instead of the previous intake of the workspace indices.

**Report to:** Jos (ISIS)

**To test:**
Run this script in MantidPlot:
``` python
OFFSPEC44956 = LoadISISNexus(Filename="OFFSPEC44956")
OFFSPEC44937 = LoadISISNexus(Filename="OFFSPEC44937")
ReflectometryReductionOneAuto(InputW [... cut]
closed 2018-11-06 by martyngigg