mantidproject/mantid #14021: Upgrade Find Reflectometry Lines

OwenArnold, 2015-10-16:
I suspect FindReflectometryLines that this algorithm is not being heavily used. So I suggest Deprecating it.

We need a new version, primarily aimed at OFFSPEC. This should be version 2, if the name above is kept, but a better name might could be used.

* Find Two maxima in the same ways as done before.
* Use the geometry setup to predict which line is going to be the Specular. Corresponds to having the same ThetaF as ThetaI. 
* Determine the width of each peak at the 0.5 max intensity of each peak.
* Return the SpectrumNumber, WorkspaceIndex, and width information for each Peak so that it can be fed into `RefectometryReductionOne` as `ProcessingInstructions` and `DirectBeamRegion`.

OwenArnold, 2015-10-16:
@raquelalvarezbanos FYI, but I'll probably fix this issue.

OwenArnold, 2015-11-19:
Proposed solution.

New algorithm called `PredictSpecular`.

Does the following:

* Crops workspace to StartWavelength, EndWavelength range
* Integrates over the full range
* Transposes the workspace so that we have a single array
* Records everything looking like a peak, but then discards all but the 3-most intense
* Loops over each peak and calculates

gemmaguest, 2020-08-06:
Closing as stale because this is not currently a priority so we are not likely to do it any time soon

closed 2020-08-06 by gemmaguest