mantidproject/mantid #14040: Load FITS: fix options and add new ones

FedeMPouzols, 2015-10-19:
Add these two options: FlipUpDown, FlipLeftRight, for convenience.

Fix the options BinsSize / FilterNoiseLevel / LoadAsRectImg when used together (currently the noise filtering is not working appropriately, and both BinSize and FilterNoiseLevel will only work when loading as one spectrum per pixel).

FedeMPouzols, 2015-10-19:
It would be convenient to add an option `AppendToGroup` and set its default to false. This would mean that if the output workspace already exists, instead of appending the new workspaces, the old ones are evicted from the group and the output group has only the newly loaded workspaces. As suggested by James Lord the current behavior can be confusing and it's not in line with what other algorithms do (example: GroupWorkspaces).

DimitarTasev, 2017-03-13:
This algorithm will be deprecated/deleted soon, https://github.com/mantidproject/mantid/issues/18242 is the only changes that will be made on the algorithm

closed 2017-03-13 by DimitarTasev