mantidproject/mantid #14049: Check cry_simplefocus with GEM data

FedeMPouzols, 2015-10-20:
Check if code like the `main` of cry_simplefocus.py works sensibly with GEM data (take the run number used there as first example: 35493). If this goes well then we should add a simple system test for it. 

This would be an ideal first point to attack when refactoring/tidying up the code.

In principle there are (commented out) examples in that file for HRPD runs, but here they don't seem to be masking any pulses which I would interpret as: this script doesn't really work appropriately for hrpd.

shahrozahmed1, 2015-10-21:
while trying to run `GEM` data with `cry_focus.focus_all()` I realised that, it requires a `GEM` specific setting `mtd.pref` and other particular files in order to work. However it is likely that once we have these settings and files, we might well be able to run `ISIS Powder script` with various instruments.

shahrozahmed1, 2015-11-09:
@FedeMPouzols now that we have found a way to run the `PowderISIS` script with `GEM`, would it be better to create a separate system test to `ISISPowderDiffractionTest` which is only testing `HRPD`?

FedeMPouzols, 2015-11-09:
Yes, that makes sense. After our meeting with Ron last week the status of this has changed. It seems that `cry_simplefocus` is not currently used.

PR #14754 (2015-12-11): removal of cry_simplefocus.py file from PowderISIS script
fixes #14049

@FedeMPouzols As this file is not required and also not being utilised anywhere in the PowderISIS script, it can be safely be removed. 

**To Test**
As long as all the PowderISIS system tests passes which include `ISISPowderDiffractionGemTest`, `ISISPowderDiffractrionHrpd` & `ISISPowderDiffractionPolarisTest`; this branch is safe to be merged.
closed 2015-12-11 by FedeMPouzols