mantidproject/mantid #14058: Refl Gui Bug

OwenArnold, 2015-10-20:
Here's the bug as reported by Max
```
I'm afraid there is quite a sever bug in the Reflectometry table in Mantid 3.5.0: it doesn't stitch the data. I have tried it on several PCs. It seems to process the two angles correctly, but as the output from the stitching it only give the high angle.
```

This relates to the older GUI, not the new one.

Lets get this fixed ASAP.

OwenArnold, 2015-10-20:
There are things wrong here. 
1) Here's the plotting output taken from running the first row from INTER_NR_test2.tbl
[image]
Notice that the stitched and high angle I(Q) plots vary only by scale. The stitched one should be much wider to incorporate the IvsQ output from the first run.

2) Also this is somehow wrong. We seem to have lost the column for the 2nd run title?
[image]

MattKing06, 2015-10-21:
```python
#Setting QMin and QMax from Data directly
Qmin = min(w1.readX(0))
Qmax = max(w2.readX(0))

#We then reset QMin and QMax based off columns
if len(self.tableMain.item(row, i * 5 + 3).text()) > 0:
    Qmin = float(self.tableMain.item(row, i + 3).text())
if len(self.tableMain.item(row, i * 5 + 4).text()) > 0:
    Qmax = float(self.tableMain.item(row, i * 5 + 4).text())

#Qmin and Qmax are passed directly into the algorithm to stitch workspaces
wcomb = combineDataMulti(wksp, outputwksp, overlapLow, overlapHigh, Qmin, Qmax, -dqq, 1, keep=True)
```

I found that setting QMin and QMax based off the qmin and qmax columns can cause this bug to occur because we can change which columns we wish to display. For example, if we have all of the columns possible present, then Qmin's value (based off column position) will actually be the value found in the QMin value for the second run, not the QMin value of the first run.

Unless this cannot be corrected for good reason, I can't see why the values would be set from the columns instead of the actual data itself. However if it cannot be changed, then perhaps a better method of extracting the correct QMin and QMax should be implemented.

PR #14069 (2015-10-22): Find correct QMin and QMax values for stitching plots in Reflectometry GUI
Fixes #14058

Previously the values for QMin and QMax were extracted from the table using an offset to find the correct column for QMin and QMax values. The iterative part of the offset was never being increased as the code for calculating the extents was outside of the for loop that the iterator was associated with.
closed 2015-10-22 by OwenArnold