mantidproject/mantid #14085: Auto grouping in the ALC interface

raquelalvarezbanos, 2015-10-22:
From James:

>I'm trying to analyse some recent CHRONUS data in the Muon ALC Analysis interface.

>I've set the default instrument to CHRONUS (just in case), browsed to the correct files for the start and end, selected a log value, etc. I put Grouping=Auto. But when I click "Load" I get an error "Couldn't load detector grouping".

>However loading the same runs into the Muon Analysis interface, it picks up the default grouping table automatically.

>Using "Custom" grouping in the ALC interface does work but I had to copy/paste the lists out of the Muon Analysis Grouping tab (typing in 600 numbers wasn't an option!)

raquelalvarezbanos, 2015-10-23:
This is not a bug in the ALC interface. If I try:

> Load(Filename=r'CHRONUS00003422.nxs', OutputWorkspace='CHRONUS00003422', DeadTimeTable='deadtable', DetectorGroupingTable='detectortable')

No grouping information will be loaded, as this information is missing in the nexus file:

[image]

There should be an entry called  "grouping", as in the following HIFI file:

[image]

The MuonAnalysis interface is able to load the grouping because when it realizes there's no grouping information it (silently, no warning message is shown) reads the instrument definition file and loads the grouping from there.

raquelalvarezbanos, 2015-10-27:
From James:

> Yes, it would be best if all three (normal Load, Muon Analysis Interface and ALC Interface) worked the same and found whatever grouping table was available.

Hence, the relevant bit of code in MuonAnalyisis must be moved to LoadMuonNexus.

tom-perkins, 2016-02-09:
The same error is seen in the case when the file **does** contain a grouping entry, but that entry is all zeros. In this case the loaded grouping is null and MuonAnalysis uses the IDF grouping.
For example:
[image]

tom-perkins, 2016-02-11:
Once PR #15191 is merged then the code from MuonAnalysis to load the grouping from the IDF will be in `API` and available for use here.

PR #16107 (2016-05-10): Load grouping from IDF if not present in file
`LoadMuonNexus` should load grouping from IDF if there is no entry in the NeXus file.

The property *DetectorGroupingTable* in `LoadMuonNexus-v1` is an output property that returns
a grouping table. If there is no grouping information in the NeXus file but this property is specified,
first try loading grouping from the IDF. If that fails, use a dummy grouping. (In either of these cases,
a log message specifies what has been done).

Also deal with files in which there is a grouping entry, but it is all zeros or otherwise invalid.

Use case: this ena [... cut]
closed 2016-05-10 by AndreiSavici