mantidproject/mantid #14088: MWRunFiles does not seem to like '-' characters in instrument names?

FedeMPouzols, 2015-10-22:
Observed while working on something else in the EnggDiffraction interface. If we pass 'ENGINX' to `setInstrumentOverride()` MWRunFiles is happy and find files that it should find. But if we pass 'ENGIN-X' it fails to find them, or at least it displays the red blob, even though we can see a log message that says that the hint is `ENGIN-X<run_no>` which looked correct and `Load` can load succesfully.

FedeMPouzols, 2015-10-22:
@shahrozahmed1 : let's keep an eye on this. But I wouldn't worry much about it. We can just use the short name of ENGINX.

FedeMPouzols, 2015-10-22:
Some of this regex matches might be misinterpreting the input string:
https://github.com/mantidproject/mantid/blob/master/MantidQt/MantidWidgets/src/MWRunFiles.cpp#L740-L757
The '-' is tricky here because it can also mean a range of run numbers, although at first sight the regexes look good and safe.

NickDraper, 2016-02-19:
As we have a workaround for now lets archive this one

closed 2016-02-19 by NickDraper