mantidproject/mantid #14099: DataProcessingAlgorithm fetching from AnalysisDataService

OwenArnold, 2015-10-23:
ReflectometryReductionOneAuto should not have to go to the ADS to get workspaces. See for example [here](https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/ReflectometryReductionOneAuto.cpp#L516) Workspaces should be extracted directly from the properties. We should not use the ADS for indirection.

Also, at the point of re-calculation, we should ensure that NO transmission correction is being applied. There's a new property to ensue that we do not do transmission correction as second time.

OwenArnold, 2015-10-30:
@raquelalvarezbanos Ok. When you get to it ask me.

MattKing06, 2016-01-18:
This has not been possible due to known issues between the ADS and WorkspaceGroups. Unfortunately we cannot access properties of child algorithms when they are workspace groups as this has not been exposed to `Algorithm` yet.

The only places that contained code accessing the ADS where `processGroups` and `checkGroups` both of which will only handle groups. So I believe, for now, this will not be achievable and the code should remain as it is.

raquelalvarezbanos, 2016-01-19:
I've seen this [here](https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/PlotAsymmetryByLogValue.cpp#L558-L559) too, where we had to use `ScopedWorkspaces` to deal with groups.

closed 2016-02-18 by NickDraper

OwenArnold, 2016-02-18:
This is a problem, but can't be fixed until we deal with the core nature of the entanglement of the processing in mantid with the ADS. This is a problem which will eventually resurface until we fix those underlying problems.