mantidproject/mantid #14120: Broken handling of workspaces in Python

SimonHeybrock, 2015-10-26:
Python variables that refer to a Mantid workspace behave in a very dangerous way: Existing workspaces with a name identical to the new variable name in Python are overwritten without warning. The existing workspace is deleted, leaving an invalid Python variable behind that still points to the deleted memory.

Example 1:
```python
from mantid.simpleapi import *

def myfunction():
    ws = CreateSampleWorkspace()

ws = CreateSampleWorkspace()
print ws.name()
myfunction()
print ws.name()
```
Example 2:
```python
from mantid.simpleapi import *

workspaces = []

for i in range(2):
    ws = CreateSampleWorkspace()
    workspaces.append(ws)

print workspaces
```
Both of the scripts crash (sometimes not, by chance).

Basically, variables that refer to a workspace cannot be used anywhere, without risking broken code:
- cannot be put in a Python container (like list)
- cannot be used in a function, because another function might use a variable of the same name
- same goes for `if` statements, etc., ...

In other words, *every* *variable* in Python that refers to a *Mantid* *workspace* is *implicitly* and without warning *lifted* *to* *global* *scope*!

A further, probably related problem is that removing a workspace from the `AnalysisDataService` in Python deletes it, even if there is still a Python variable referring to it (contrary to the behavior in C++!).

SimonHeybrock, 2015-11-02:
Another example, the issues are not restricted to using a Python workspace variable on the left-hand side of an assignment, but also as a parameter to an Algorithm:

```python
from mantid.simpleapi import *

ws_in = CreateSampleWorkspace()
ws = CreateSampleWorkspace()
print ws.name()
Rebin(InputWorkspace=ws_in, OutputWorkspace=ws, Params='0.1,0.1,1')
print ws.name()
```
The error is sth. like:
```python
Traceback (most recent call last):
  File "test3.py", line 7, in <module>
    print ws.name()
RuntimeError: Variable invalidated, data has been deleted.
```

OwenArnold, 2015-11-04:
@SimonHeybrock This is a known problem, without a simple solution. Algorithms (except those run with the child flag) in Mantid always have a side-effect. That side-effect is to put the outputs in a dictionary, and that leads to naming conflicts, simulated race conditions and some other nasty behaviour when it comes to GroupWorkspaces. 

That being said, removing the implicit dictionary entries upon algorithm completion would be a major change to the operation of Mantid, and probably an unpopular one with users.

OwenArnold, 2015-11-04:
@SimonHeybrock. I haven't thought this through in detail, but one possibility may be to treat the current behaviour as a *legacy* mode, and introduce a *future* mode which provides safety around automatic replacement of dictionary entries. In the *future* mode, replacement of existing keys would raise an exception at the exact point of failure. Users would be able to explicitly replace the entry, but it would have to be consciously done.

OwenArnold, 2015-11-04:
@SimonHeybrock @martyngigg Another solution might be to treat the key entries in the ADS as private, and unique in some way, and have the workspace names as aliases to those keys. We could then support duplicate entries, or at least two entries that appear to have the same name. Instead of failing with a segfault, we could then potentially spot the duplication and raise some kind of Ambiguity exception at the point that the workspace as accessed.

martyngigg, 2015-11-04:
The segfault actually shouldn't happen so that is a bug. I would have expected the `Variable invalidated` exception.

NickDraper, 2015-11-05:
Removed the release and high priority labels, as it is not clear that this is a bug that needs urgent fixing, rather than an understanding of how Mantid works.

@OwenArnold Allowing Duplicate keys within the ADS seems like a great recipe for confusion.  They only genuine use case for this would be in a multi user environment, but we would either adjust the ADS to handle user separation, or have each user have their own ADS.

NickDraper, 2016-02-19:
We need to agree where to go with this

OwenArnold, 2016-02-19:
@NickDraper I think that @SimonHeybrock prototype is the only correct way to handle this. He has an implementation of this called `mantid.experimental.pythonapi` [here](dependence https://github.com/mantidproject/mantid/pull/14686 ). Eventually It'd be good to see the ADS dependence in this completely removed.

closed 2021-03-01 by stale[bot]