mantidproject/mantid #14122: Problems with ConvertToReflectometryQ

OwenArnold, 2015-10-27:
There is likely to be a problem with the way that `ConvertToReflectometryQ` is performing it's calculations. We end up seeing fringes NaNs around regions of interest in the plotted data.

The script below contains a hand-rolled function that converts the Theta-Lambda map into a QxQy map. For the same data it also runs ConvertToReflectometryQ, which should be performing the same operation.

We need to understand why the results are coming out differently, because the basic calculation is very simple. Once the fix has been determined. I suggest you write another issue to perform the fix, and assign it to Matt King.

The input data is from OFFSPEC `36265+36266+36267+36268`.

Use Owen to help with this issue.

```

def doDSSCorrections(wksp,angle=1.2,nper=0,Nqx=200,Nqz=200):
    # get the lambda and theta arrays from the original data
    thetaf=[]
    if nper == 0:
        suffix1='detnorm'
        suffix2='qxqz'
        suffix3='qxlam'
    else:
        suffix1='detnorm_'+str(nper)
        suffix2='qxqz_'+str(nper)
        suffix3='qxlam_'+str(nper)
    a1=mtd[wksp+suffix1]
    nth=a1.getNumberHistograms()
    ntc=len(a1.dataY(0))
    thetaf=a1.getAxis(1).extractValues()
    thetaf=thetaf*pi/180.0
    lambda0=a1.getAxis(0).extractValues()
    lambda1=[]
    for i in range(len(lambda0)-1):
        lambda1.append(0.5*(lambda0[i]+lambda0[i+1]))
    dthf=float(nth-1)/(thetaf[-1]-thetaf[0])
    dlam=float(ntc-1)/(lambda1[-1]-lambda1[0])
    # get the qx and qz arrays from the data we just created
    a2=mtd[wksp+suffix2]
    lmin=lambda0[0]
    lmax=lambda0[-1]
    lamstep=(lmax-lmin)/(Nqz-1)
    lam2=[]
    for i in range(Nqz):
        lam2.append(lmin+i*lamstep)
    qz=a2.getAxis(1).extractValues()
    qx=a2.getAxis(0).extractValues()
    cthetai=cos(angle*pi/180.0)
    sthetai=sin(angle*pi/180.0)
    thetai=angle*pi/180.0
    thetaf0=thetaf[0]
    lambda0=lambda1[0]
    for i in range(Nqz):
        qzi=qz[i]
        #qzi=lam2[i]
        pi2qz=2.0*pi/qzi
        for j in range(Nqx):
            qxj=qx[j]
            ang=(qzi*cthetai-qxj*sthetai)/sqrt(qzi*qzi+qxj*qxj)
            #ang=cthetai-(qxj*qzi/(2.0*pi))
            ang=min(1.0,ang)
            ang=asin(ang)
            #ang=acos(ang)
            if qxj==0.0:
                ang=thetai
            else:
                ang=pi-ang-atan(qzi/qxj)
                if ang > pi:
                    ang=ang-pi
            lam=pi2qz*(sthetai+sin(ang))
            #lam=qzi
            xind=(ang-thetaf0)*dthf
            yind=(lam-lambda0)*dlam
            indy=int(yind)
            indx=int(xind)
            if indy >=  0 and indy <= ntc-2 and indx >= 0 and indx <= nth-2:
                dyind=yind-float(indy)
                dxind=xind-float(indx)
                ofsp00=a1.dataY(indx)[indy]
                ofsp01=a1.dataY(indx)[indy+1]
                ofsp10=a1.dataY(indx+1)[indy]
                ofsp11=a1.dataY(indx+1)[indy+1]
                offsp1=(1.0-dxind)*ofsp00+dxind*ofsp10
                offsp2=(1.0-dxind)*ofsp01+dxind*ofsp11
                a2.dataY(i)[j]=(1.0-dyind)*offsp1+dyind*offsp2
                ofsp00=a1.dataE(indx)[indy]
                ofsp00=ofsp00*ofsp00
                ofsp01=a1.dataE(indx)[indy+1]
                ofsp01=ofsp01*ofsp01
                ofsp10=a1.dataE(indx+1)[indy]
                ofsp10=ofsp10*ofsp10
                ofsp11=a1.dataE(indx+1)[indy+1]
                ofsp11=ofsp11*ofsp11
                offsp1=((1.0-dxind)*(1.0-dxind))*ofsp00+(dxind*dxind)*ofsp10
                offsp2=((1.0-dxind)*(1.0-dxind))*ofsp01+(dxind*dxind)*ofsp11
                a2.dataE(i)[j]=sqrt(abs((1.0-dyind)*(1.0-dyind)*offsp1+dyind*dyind*offsp2))
            else:
                a2.dataY(i)[j]=0.0
                a2.dataE(i)[j]=0.0
    w1=mtd[wksp+suffix2]*1.0
    w2=mtd[wksp+suffix2]
    a2=w1.getAxis(1)

def DSqxqz(run1,wksp,angle=1.2,qxqzlimits='-5e-4,5e-4,0.02,0.1',binning1=["1.5","0.02","14.0","2"],Nqx=200,Nqz=200,withpol=1):
    halftheta = angle/2.0
    #binning1=["1.0","0.05","14.0","2"]
    if withpol==1:
        nr.nrPNRFn(run1,wksp,str(halftheta),'none',"404","404","404",binning1,"",'0',['2','1'],'0',dofloodnorm=2)
    else:
        nr.nrNRFn(run1,wksp,str(halftheta),'none',"404","404","404",binning1,"",dofloodnorm=2)
    # Delete the norm and RvQ workspaces as they have the wrong angle
    DeleteWorkspace(wksp+'RvQ')
    DeleteWorkspace(wksp+'norm')
    ConvertSpectrumAxis(InputWorkspace=wksp+"detnorm", OutputWorkspace=wksp+"detnorm", Target='SignedTheta')
    ConvertToReflectometryQ(InputWorkspace=wksp+'detnorm', OverrideIncidentTheta=True, IncidentTheta=angle, Extents=qxqzlimits, OutputAsMDWorkspace=False, OutputWorkspace=wksp+"qxqz", NumberBinsQx=Nqx, NumberBinsQz=Nqz, OutputVertexes='_ignoreit')
    if withpol == 1:
        doDSSCorrections(wksp,angle,1,Nqx=Nqx,Nqz=Nqz)
        doDSSCorrections(wksp,angle,2,Nqx=Nqx,Nqz=Nqz)
    else:
        doDSSCorrections(wksp,angle,0,Nqx=Nqx,Nqz=Nqz)
        
        
        
        

DSqxqz("36265+36266+36267+36268","old",angle=0.64,qxqzlimits='-3e-4,4e-4,0.008,0.15',binning1=["1.0","0.025","14.0","2"],Nqx=500,Nqz=500,withpol=0)

```

raquelalvarezbanos, 2015-10-28:
To run the script you just have to add

`import offspec as nr`

at the beginning. `offspec.py` can be found in the script repository. You'll also need to download `reflectometry/offspec/WSF_Flood.nxs` and then add the corresponding path to Mantid -> Preferences -> Mantid -> Directories.

However, when running the script it fails at `ConvertSpectrumAxis` with error:

> IndexError: All arguments to init must be positive and non-zero

The reason is that the workspace we are trying to convert (called `olddetnorm`) is empty (all counts are zero). This happens because the input runs (36265, 36266, 36267, 26268) have zero counts except for the monitors, but these have been previously removed to create `olddetnorm`.

raquelalvarezbanos, 2015-10-28:
After adding the following changes the script successfully runs:

- From math import: `from math import pi, cos, sin, sqrt, asin, atan`
- The IDF must be updated, I've used `OFFSPEC_Definition_new2.xml` with: `valid-from   ="2015-10-01 22:00:00"
valid-to     ="2100-01-31 23:59:59"
last-modified="2015-10-01 15:00:00"`

OwenArnold, 2015-10-28:
FYI [here](https://github.com/mantidproject/mantid/blob/master/Framework/DataObjects/inc/MantidDataObjects/CalculateReflectometryQxQz.h) is where the calculations to get Qx, Qy are being performed for ConvertToReflectometryQ. These should be similar to those performed in the script attached to this issue.

raquelalvarezbanos, 2015-10-28:
`oldqxqz` matches `ws1`, as it is being modified in `doDSSCorrections`. So we need to clone `oldqxqx` right after `ConvertToReflectometryQ`. This is the modified script I am going to use:

```
import offspec as nr
from math import pi, cos, sin, sqrt, asin, atan

def doDSSCorrections(wksp,angle=1.2,nper=0,Nqx=200,Nqz=200):
    # get the lambda and theta arrays from the original data
    thetaf=[]
    if nper == 0:
        suffix1='detnorm'
        suffix2='qxqz'
        suffix3='qxlam'
    else:
        suffix1='detnorm_'+str(nper)
        suffix2='qxqz_'+str(nper)
        suffix3='qxlam_'+str(nper)
    a1=mtd[wksp+suffix1]
    nth=a1.getNumberHistograms()
    ntc=len(a1.dataY(0))
    thetaf=a1.getAxis(1).extractValues()
    thetaf=thetaf*pi/180.0
    lambda0=a1.getAxis(0).extractValues()
    lambda1=[]
    for i in range(len(lambda0)-1):
        lambda1.append(0.5*(lambda0[i]+lambda0[i+1]))
    dthf=float(nth-1)/(thetaf[-1]-thetaf[0])
    dlam=float(ntc-1)/(lambda1[-1]-lambda1[0])
    # get the qx and qz arrays from the data we just created
    a2=mtd[wksp+suffix2]
    lmin=lambda0[0]
    lmax=lambda0[-1]
    lamstep=(lmax-lmin)/(Nqz-1)
    lam2=[]
    for i in range(Nqz):
        lam2.append(lmin+i*lamstep)
    qz=a2.getAxis(1).extractValues()
    qx=a2.getAxis(0).extractValues()
    cthetai=cos(angle*pi/180.0)
    sthetai=sin(angle*pi/180.0)
    thetai=angle*pi/180.0
    thetaf0=thetaf[0]
    lambda0=lambda1[0]
    for i in range(Nqz):
        qzi=qz[i]
        #qzi=lam2[i]
        pi2qz=2.0*pi/qzi
        for j in range(Nqx):
            qxj=qx[j]
            ang=(qzi*cthetai-qxj*sthetai)/sqrt(qzi*qzi+qxj*qxj)
            #ang=cthetai-(qxj*qzi/(2.0*pi))
            ang=min(1.0,ang)
            ang=asin(ang)
            #ang=acos(ang)
            if qxj==0.0:
                ang=thetai
            else:
                ang=pi-ang-atan(qzi/qxj)
                if ang > pi:
                    ang=ang-pi
            lam=pi2qz*(sthetai+sin(ang))
            #lam=qzi
            xind=(ang-thetaf0)*dthf
            yind=(lam-lambda0)*dlam
            indy=int(yind)
            indx=int(xind)
            if indy >=  0 and indy <= ntc-2 and indx >= 0 and indx <= nth-2:
                dyind=yind-float(indy)
                dxind=xind-float(indx)
                ofsp00=a1.dataY(indx)[indy]
                ofsp01=a1.dataY(indx)[indy+1]
                ofsp10=a1.dataY(indx+1)[indy]
                ofsp11=a1.dataY(indx+1)[indy+1]
                offsp1=(1.0-dxind)*ofsp00+dxind*ofsp10
                offsp2=(1.0-dxind)*ofsp01+dxind*ofsp11
                a2.dataY(i)[j]=(1.0-dyind)*offsp1+dyind*offsp2
                ofsp00=a1.dataE(indx)[indy]
                ofsp00=ofsp00*ofsp00
                ofsp01=a1.dataE(indx)[indy+1]
                ofsp01=ofsp01*ofsp01
                ofsp10=a1.dataE(indx+1)[indy]
                ofsp10=ofsp10*ofsp10
                ofsp11=a1.dataE(indx+1)[indy+1]
                ofsp11=ofsp11*ofsp11
                offsp1=((1.0-dxind)*(1.0-dxind))*ofsp00+(dxind*dxind)*ofsp10
                offsp2=((1.0-dxind)*(1.0-dxind))*ofsp01+(dxind*dxind)*ofsp11
                a2.dataE(i)[j]=sqrt(abs((1.0-dyind)*(1.0-dyind)*offsp1+dyind*dyind*offsp2))
            else:
                a2.dataY(i)[j]=0.0
                a2.dataE(i)[j]=0.0
    w1=mtd[wksp+suffix2]*1.0
    w2=mtd[wksp+suffix2]
    a2=w1.getAxis(1)

def DSqxqz(run1,wksp,angle=1.2,qxqzlimits='-5e-4,5e-4,0.02,0.1',binning1=["1.5","0.02","14.0","2"],Nqx=200,Nqz=200,withpol=1):
    halftheta = angle/2.0
    #binning1=["1.0","0.05","14.0","2"]
    if withpol==1:
        nr.nrPNRFn(run1,wksp,str(halftheta),'none',"404","404","404",binning1,"",'0',['2','1'],'0',dofloodnorm=2)
    else:
        nr.nrNRFn(run1,wksp,str(halftheta),'none',"404","404","404",binning1,"",dofloodnorm=2)
    # Delete the norm and RvQ workspaces as they have the wrong angle
    DeleteWorkspace(wksp+'RvQ')
    DeleteWorkspace(wksp+'norm')
    ConvertSpectrumAxis(InputWorkspace=wksp+"detnorm", OutputWorkspace=wksp+"detnorm", Target='SignedTheta')
    ConvertToReflectometryQ(InputWorkspace=wksp+'detnorm', OverrideIncidentTheta=True, IncidentTheta=angle, Extents=qxqzlimits, OutputAsMDWorkspace=False, OutputWorkspace=wksp+"qxqz", NumberBinsQx=Nqx, NumberBinsQz=Nqz, OutputVertexes='_ignoreit')
    CloneWorkspace(InputWorkspace=wksp+'qxqz',OutputWorkspace=wksp+'qxqzCopy')
    if withpol == 1:
        doDSSCorrections(wksp,angle,1,Nqx=Nqx,Nqz=Nqz)
        doDSSCorrections(wksp,angle,2,Nqx=Nqx,Nqz=Nqz)
    else:
        doDSSCorrections(wksp,angle,0,Nqx=Nqx,Nqz=Nqz)

DSqxqz("36265+36266+36267+36268","old",angle=0.64,qxqzlimits='-3e-4,4e-4,0.008,0.15',binning1=["1.0","0.025","14.0","2"],Nqx=500,Nqz=500,withpol=0)
ws1 = mtd['oldqxqzCopy']
ws2 = mtd['w1']
result = CheckWorkspacesMatch(ws1,ws2)
print result
```

raquelalvarezbanos, 2016-01-14:
This is no longer an issue. It turned out that scientists were using an approximation in their script, which explained why our results were coming out differently. Since then, we have identified and fixed a number of problems related to `ConvertToReflectometryQ` (#14297 #14477 and #14371), which seems to be working properly now.

OwenArnold, 2016-01-14:
@raquelalvarezbanos. Agreed, and OK for this to be closed but there does seem to be another problem with overlaps that Jos has reported. You might need to ask him to demo that.

raquelalvarezbanos, 2016-01-14:
@OwenArnold when did he report this new problem, and, is there an issue for it?

closed 2016-01-22 by NickDraper