mantidproject/mantid #14176: Logging to console does not get more detailed than Notice

peterfpeterson, 2015-10-29:
To reproduce add the following lines to your `.Mantid.user.properties` then run any algorithm.

```
logging.loggers.root.level=information
logging.loggers.fileFilterChannel.level=information
logging.loggers.consoleFilterChannel.level=information
```

The log file will have things at information, but the console will not.

NickDraper, 2015-11-02:
Example is wrong and that is the problem

For the two specific channels it is logging.channels.xxx not logging.loggers.xxx

NickDraper, 2015-11-02:
So this works in the user.properties file

logging.loggers.root.level=debug
logging.channels.consoleFilterChannel.level= debug

NickDraper, 2015-11-02:
No code changes - ready for review

peterfpeterson, 2015-11-02:
Works for me with these changes.

closed 2015-11-02 by peterfpeterson