mantidproject/mantid #14193: Propergate Transfer Issues to Surface in Reflectometry GUI

OwenArnold, 2015-10-30:
We recently added a new Mode to transfer files based on MeasurementId in the [Reflectometry GUI](http://docs.mantidproject.org/nightly/interfaces/ISIS_Reflectometry.html). Some finishing work needs to be done around this. See [here](https://github.com/mantidproject/mantid/pull/14051) for the last set of changes.

Make sub-issues for each of the following

1. The code in CustomInterfaces relating to the Refl gui should be moved into a subfolder like it is for the other GUIs
1. The `Measurement` type needs to be renamed to renamed `MeasurementRun` because conceptually a group of `MeasurementRuns` make up a `Measurement`. Likewise `MeasurementSource` should be renamed `MeasurementRunSource`.
1. The `Measurement` records error information regarding any impossible transfer. We should set the column in the UI to highlight the failed transfer and record this information. I was thinking that the row in the transfer table should be highlighted orange, and should provide some way of explaining why the transfer failed.
1. Would be nice if the transfer could be made async via the AlgorithmRunner.

OwenArnold, 2015-11-03:
@tom-perkins better that @MattKing06  does this on reflection.

MattKing06, 2015-11-05:
Blocked by #14274

gemmaguest, 2019-07-24:
I think the remaining tasks are done as part of the GUI restructuring but there may be uses of AlgorithmManager that can be tidied up more - I've added that to #26025

closed 2019-07-24 by gemmaguest