mantidproject/mantid #14203: ChangeOfBasisMatrix not being set for MDHistoWorkspace with nonorthogonal axes

quantumsteve, 2015-10-30:
Today I opened a MDHistoWorkspace with hexagonal coordinates in the VSI and the axes were not displayed correctly. The ChangeOfBasisMatrix was set to the identity matrix. Working backwards, the issue occurs at [L145-146 of vtkDataSetToNonOrthogonalDataSet.cpp](https://github.com/mantidproject/mantid/blob/master/Vates/VatesAPI/src/vtkDataSetToNonOrthogonalDataSet.cpp#L145) where m_coordType is set to `none` and an exception is thrown on the next line.

This appears to be a side effect of pull request #13917, which made significant changes to `MDHistoWorkspace::getSpecialCoordinateSystem()`. 

@AndreiSavici checked that this appears to only occur on older MDHistoWorkspaces and NOT ones generated by during the same session. 

A unit test should also be added to verify that the ChangeOfBasisMatrix is set correctly.

quantumsteve, 2015-10-30:
Commenting out [L145-149 of vtkDataSetToNonOrthogonalDataSet.cpp](https://github.com/mantidproject/mantid/blob/master/Vates/VatesAPI/src/vtkDataSetToNonOrthogonalDataSet.cpp#L145) restores the previous behavior on this file, but probably breaks other things.

picatess, 2015-11-02:
@quantumsteve Could you send me your test file? I will have a look at this.

AndreiSavici, 2015-11-02:
https://www.dropbox.com/s/slytdzecam4byti/test_non_orthogonal.nxs?dl=0

PR #14312 (2015-11-10): Fix load md for legacy files
Fixes #14203

#For Testing 

* Code reivew
* Make sure that the new unit tests pass
* Load a legacy file with non-ortho axes into the VSI (e.g. symm.nxs)
  * Confirm that it is displayed with non-ortho axes
closed 2015-11-10 by quantumsteve