mantidproject/mantid #14209: ConvexPolygon assignable behaviour

OwenArnold, 2015-11-02:
As a corollary to fix #14200 I notice that the use-case for *FractionalRebinning* [here](https://github.com/mantidproject/mantid/blob/master/Framework/DataObjects/src/FractionalRebinning.cpp#L147) has a single ConvexPolygon object, which is global to the loop, and is *cleared* via `clear` and then inserted into via `insert` upon each iteration. It raises a few questions:

1. Surely you have n assignments more than necessary because clear is setting all members to double max, double min etc?
1. This section of the FractionalRebinning code does not look like it can be made thread-safe in it's current form. Is that why we have a parallel critical block, but no openmp loop?
1. Given that the solution to both of the above would be to make ConvexPolygon immutable, is there any use-case I've missed?

@martyngigg since you seem to be the original author of this, perhaps you could comment on the questions and then reassign to @tom-perkins for the fix.

martyngigg, 2015-11-02:
1. It's unfortunate I didn't put a better comment in the code but I think I remember observing that the internal vertex list reused the same memory so even though the number of assignments is higher it was avoiding dynamically allocating the new vector with each loop as clear didn't seem to throw the memory away completely.
2.  The code in it's original form was in the `Rebin2D` algorithm that does contain an `omp` parallel loop - https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/Rebin2D.cpp#L118. The critical block is required to protect the workspace write access. The code refactor has meant that it's just now not as obvious that there is a higher-level loop running this method in parallel.

It's not obvious what level the parallelism is best applied. It feels like the current behaviour of parallelism over the outermost loop is correct and the `rebinToOutput` is effectively a task for the threads to run. This should all be informed by profiling though I think. @OwenArnold Is this still something you think is worth looking at?

OwenArnold, 2015-11-02:
@martyngigg. I don't think that this is high priority, but I think it would be better if possible to let the compliers figure out whether memory can be reused rather than forcing it ourselves if possible. Some variant on the current Performance test for [ConvexPolygonTest](https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/test/ConvexPolygonTest.h#L166:L173) could be used as a cross-platform benchmark. I think that the immutable semantics would be desirable as it would give the thread-safety and provide more options on how the parallelism could occur over a ConvexPolygon. Clearly if there is a large observed performance hit to removing the overwrite behaviour that would trump any of the concerns I have listed.

closed 2021-03-03 by stale[bot]