mantidproject/mantid #14249: Workspace groups: progress bar doesn't show any progress

FedeMPouzols, 2015-11-03:
This happens for example with `Rebin` or `SumSpectra`. If you run it on individual workspaces the progress bar works well. If you run the algorithms on a group the progress bar is created but it doesn't move at all.

For example any event and multi-period workspace should reproduce this issue. One particular example:
From \\isis\inst$\NDXENGINX\Instrument\data\cycle_12_4, ENGINX00197019.NXS

This was initially (and wrongly) identified as an issue with `RebinByPulseTimes` but it seems to happen in general with workspace groups.

We've seen this happening on linux and windows, with @LamarMoore .

@martyngigg : I just created this more general issue.

NickDraper, 2016-06-13:
Test script
``` python 
ws1=CreateSampleWorkspace(numBanks=100)
ws2=CloneWorkspace(ws1)
ws3=CloneWorkspace(ws1)
ws4=CloneWorkspace(ws1)
wsGroup1=GroupWorkspaces("ws1,ws2,ws3,ws4")

wsResult=Rebin(wsGroup1,"5")
```

NickDraper, 2016-06-13:
test script for MergeRuns (special case)
```python
ws_1=CreateSampleWorkspace(numBanks=100)
AddSampleLog(ws_1,"nperiods","4")
ws_2=CloneWorkspace(ws_1)
ws_3=CloneWorkspace(ws_1)
ws_4=CloneWorkspace(ws_1)
AddSampleLog(ws_1,"current_period","1")
AddSampleLog(ws_2,"current_period","2")
AddSampleLog(ws_3,"current_period","3")
AddSampleLog(ws_4,"current_period","4")
ws=GroupWorkspaces("ws_1,ws_2,ws_3,ws_4")

wsResult=MergeRuns("ws,ws,ws")
```

PR #16606 (2016-06-27): Progress reporting for algs with group workspaces
Changed the group workspace handling of algorithms to route the dynamically created workspace progress reporting through the parent algorithm, correctly scaling on the way.

MergeRuns is a special case that is hadled seperately, but this has been done as well.

**To test:**

 1. Run some algorithms with group workspaces.  They should work at least as well as the underlying algorithm reporting does.
 1. look at the underlying issue, that has test python scripts for Rebin, and MergeRuns in the comments #14249 

Fixes #14249 .

RELEASE NOTES
docs/ [... cut]
closed 2016-06-27 by peterfpeterson