mantidproject/mantid #14250: Engg::valid

shahrozahmed1, 2015-11-03:
Now that the event mode is supported by the Engineering Diffraction Interface and the  basic pre-processing (re-binning) capability is implemented. Right now the nexus file `ENGINX00236516` is provided, which has with more than 14000 Bins and roughly 15mb, it seems to be eating up lot of computer memory and freezes the computer for half an hour on Windows while preforming the `Rebin` under `Pre-processing` tab. This issue can explained to Saurabh and he would be able to tell if this issue is a high propriety and what are the average/ideal size files that are currently being used.

FedeMPouzols, 2015-11-03:
Yes, we have to see what to do about this. It seems that the files are going to be big enough that it is too easy for the user to use a too small bin size (by mistake or lack of awareness) which can require allocating huge amounts of memory and (if you're able to allocate all) then run a very lengthy process trashing to disk all the time.

FedeMPouzols, 2016-08-12:
My guess would be that this was happening because the combination of workspace size (biggish) and bin size (tinyish) was producing a large dataset, so the system was trashing. Maybe a warning could be added to tell the users in advance what could happen. We have some code in Mantid that will estimate your memory resources.

FedeMPouzols, 2016-08-24:
@DavidFair: I think that this is the same issue or very closely related to what you experienced when rebinning the split files from the event dataset from Manchester.

DavidFair, 2016-08-25:
@FedeMPouzols I would guess so - it looks very similar to what happens if you perform rebin without a start and end parameter. We could easily put a check in place for it however is this an issue with Rebin or the interface? As I think the interface just passes the params directly to Rebin.

DavidFair, 2016-09-13:
Following up on this it turns out we can limit the data between 0.5-4 Angstroms.
In TOF this is multiplied by 18400 (average difC value) giving 9200-73600 as the start and end range in TOF for an event workspace. 

Without this Rebinning will not end with the bins getting infinitely smaller as x tends to 0 which explains the high memory usage.

closed 2016-09-13 by DavidFair