mantidproject/mantid #14287: Follow up work for ConvertToReflectometryQ

OwenArnold, 2015-11-05:
Following fix to #14279

*  Make sure the polygon ordering fix is applied to the kikf and pipf modes. Currently done for the QxQy mode but not for the others.
* See the refactoring steps against the original issue PR. Apply those.
* In the documentation for the algorithm, Tim wants the theta-lambda plot that you can get from the first example included as an image. Something like this.
[image]

OwenArnold, 2015-11-05:
@raquelalvarezbanos Can you decide if you're going to do any of this. Otherwise I think Matt would be more than capable of making these fixes.

raquelalvarezbanos, 2015-11-05:
Hi @OwenArnold I agree Matt can make the fixes. I can help if you need help @MattKing06, just let me know

PR #14455 (2015-11-17): Apply fixes in CalculateReflectometryQxQz to other transformation types
To Test:
======

**Functional Tests**
- You may need [this file](https://slack-files.com/T02J4MMEW-F0AA5NGSY-405b51aad0) to complete this test.
- Run the second [usage example](http://docs.mantidproject.org/nightly/algorithms/ConvertToReflectometryQ-v1.html) for ConvertToReflectometryQ.
- This should yield your `vertexes_kikf`,`vertexes_pipf` and, `vertexes_qxqz` workspaces.
- I have adapted the patch plot usage examples to work with each transformation and those scripts can be found here:

    [kikf_patchplot](https://sla [... cut]
closed 2015-11-17 by ElliotAOram