mantidproject/mantid #14306: ComponentAssembly Bug

OwenArnold, 2015-11-05:
```python
ws= LoadEmptyInstrument(Filename='OFFSPEC_Definition_new2.xml')
print ws.getInstrument().getComponentByName('WLSFDetector').getPos()
MoveInstrumentComponent(Workspace=ws, ComponentName='DetectorBench', Y=0.1, Z=3.62, RelativePosition=False)
print ws.getInstrument().getComponentByName('WLSFDetector').getPos()
```

WLSFDetector is a subcomponent of the DetectorBench (see attached IDF). Shifting the bench vertically has seemed to result in WLSFDetector being translated in the beam direction Z. We have traced this through to [here in component.cpp](https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/src/Instrument/Component.cpp#L339). The reason that the Z position is different is that the += is adding the original offset to itself that's why it's twice it's original value.

This is potentially quite a serious bug

[OFFSPEC_Definition_new2.xml.txt](https://github.com/mantidproject/mantid/files/28033/OFFSPEC_Definition_new2.xml.txt)

OwenArnold, 2015-11-05:
@martyngigg FYI

NickDraper, 2015-11-18:
This is not a bug, just a misunderstanding.  The IDF had the detector bench defined at 0,0,0, which is a bit strange as the detectors on it are over 3m away.  The original script included an absolute translation of the table of over 3m in Z which was the detector distance from the sample, as this was an absolute change for the table at 0,0,0, it moved the detectors attached to it a further 3m back (unintentionally).

This was verified by the attached unit test and the following python.
``` python
ws= LoadEmptyInstrument(Filename='C:/MantidInstall/instrument/OFFSPEC_Definition_new2.xml')
print 'DetectorBench', ws.getInstrument().getComponentByName('DetectorBench').getPos()
print 'WLSFDetector', ws.getInstrument().getComponentByName('WLSFDetector').getPos()
MoveInstrumentComponent(Workspace=ws, ComponentName='DetectorBench', Y=0.1, Z=3.62, RelativePosition=False)
print 'WLSFDetector', ws.getInstrument().getComponentByName('WLSFDetector').getPos()
print 'DetectorBench', ws.getInstrument().getComponentByName('DetectorBench').getPos()
MoveInstrumentComponent(Workspace=ws, ComponentName='DetectorBench', Y=0.1, Z=3.62, RelativePosition=True)
print 'WLSFDetector', ws.getInstrument().getComponentByName('WLSFDetector').getPos()
print 'DetectorBench', ws.getInstrument().getComponentByName('DetectorBench').getPos()
```

This is the correct behaviour

martyngigg, 2015-11-18:
I think the way that translations and rotations are applied needs to made much clearer in the documentation for `MoveInstrumentComponent` & `RotateInstrumentComponent`. The current documentation is very short and this is quite a complicated subject. Following this discussion it feels like we at least need a section on how shifts work when applied to component assemblies, probably with pictures to aid the explanation.

OwenArnold, 2015-11-18:
@martyngigg I think that would be a nice addition to make here.

OwenArnold, 2015-11-24:
@NickDraper could you have a look at martyn's comments.

NickDraper, 2015-11-24:
I agree, I just haven't had the chance yet!

PR #14487 (2015-12-04): Add unit test to ensure mutliple comp assemblies move correctly
The issue was an investigation, but I added a new unit test as part of it and we should keep that.

Also improved documentation of MoveInstrumentComponent and RotateInstrumentComponent

closes #14306

### release notes
http://www.mantidproject.org/ReleaseNotes_3_6_UI_Changes#Documentation

### To test
If the unit + doc tests pass all is good
closed 2015-12-04 by OwenArnold