mantidproject/mantid #14322: Investigate Removal of Data Search Directories

OwenArnold, 2015-11-06:
I've seen an issue a couple of times on the Mac now. When you reinstall a new version of Mantid, it clears the data search directories entries. Does this affect other platforms?

tom-perkins, 2015-11-06:
Tested on Windows 7 64-bit:
- Installed 3.5.0 over the top of existing 3.5.0. Data search directories were retained ok.
- Upgraded 3.5.0 to latest nightly 3.5.20151105.1804. OK
- Downgraded to 3.4.1. OK
- Started from 3.4.1, upgraded to 3.5.0. OK

OwenArnold, 2015-11-06:
@tom-perkins Can you remote into the OSX test machine. Martyn can give you login details. Try there too.

tom-perkins, 2016-10-14:
@OwenArnold @samueljackson92 Have either of you noticed this on the Mac with recent releases?

martyngigg, 2020-04-27:
This is no longer an issue after switching to the drag and drop packaging.

closed 2020-04-27 by martyngigg