mantidproject/mantid #14327: Text Axis not loaded properly from processed file

martyngigg, 2015-11-06:
To reproduce:

* Make sure you have the system test data
* Start MantidPlot
* Run LoadNexusProcessed and `VesuvioFittingWithQuadraticBackgroundTest_V3.nxs` file in `build/ExternalData/Testing/SystemTests/tests/analysis/reference/`
* Double-click the workspace to show the matrix
* The last axis label will have some random characters in it

martyngigg, 2015-11-06:
Thinking about it it's also possible this is a problem with the saving side. It's worth checking. The writing code is in https://github.com/mantidproject/mantid/blob/master/Framework/Nexus/src/NexusFileIO.cpp#L282 and the reading code is in https://github.com/mantidproject/mantid/blob/master/Framework/DataHandling/src/LoadNexusProcessed.cpp#L1674

picatess, 2015-11-06:
@OwenArnold  Don't know if this helps: This is a workspace I recently replaced. It is linked to this PR https://github.com/mantidproject/mantid/pull/13885 .The main difference to the previous system test data is the distribution flag which should be now set to true. Also, this is a very sensitive system test, the new system test data had to be generated on Ubuntu.

martyngigg, 2015-11-06:
@picatess @OwenArnold This is definitely just a bug with loading the text axes in some way. The data itself seems to compare just fine.

OwenArnold, 2015-11-06:
@picatess. Sorry to do this, but i've got to sort some things out for ANSTO immediately. Would you mind taking a look at this.

PR #14338 (2015-11-06): Fix null termination character issue when loading
Fixes  #14327 

# For testing

Make sure that the system tests pass

Run the script below and make sure to insert the correct paths. one is to the system test data and the other is needed to save and reload a temporary file. The prints should all state Success
```python
WS_PREFIX="fit"

def do_fit_with_quadratic_background():
    """
    Run the Vesuvio fit without background. If k_is_free is False then it is fixed to f0.Width*sqrt(2)/12
    """
    function_str = \
        "composite=ComptonScatteringCountRate,NumDeriv=1,IntensityConstraints=\ [... cut]
closed 2015-11-06 by martyngigg