mantidproject/mantid #14344: Easy Misuse of SansWideAngleCorrection

OwenArnold, 2015-11-08:
[SansWideAngleCorrection](http://docs.mantidproject.org/nightly/algorithms/SANSWideAngleCorrection-v1.html) Allows you to provide a transmission workspace that has not been collapsed down to a single spectra. I [this](https://github.com/mantidproject/mantid/blob/master/Framework/PythonInterface/plugins/algorithms/SANSWideAngleCorrection.py#L56) is done on the assumption that there is a single spectra to process.

I also notice that:
* Validation is being done inside execute rather than in validateInputs overrides. 
* dataY, and such unsafe variants are being called on input workspaces. Not sure this is desired.

OwenArnold, 2015-11-08:
@picatess some easy fixes to make the SANS reduction area easier to work with, and harder to misuse.

Mantid-Matthew, 2020-09-01:
Closing as stale.

closed 2020-09-01 by Mantid-Matthew