mantidproject/mantid #14349: Bounding box for hexahedron shape does not work

Anders-Markvardsen, 2015-11-09:
We recommend [to use boundingbox for hexahedron shape](http://docs.mantidproject.org/nightly/concepts/HowToDefineGeometricShape.html#howtodefinegeometricshape). 

A good example why is demonstrated by 

1. Run LoadEmptyInstrument with the IDF:
[templateSANS_Mantid.instr_without_boundingBox.txt](https://github.com/mantidproject/mantid/files/29820/templateSANS_Mantid.instr_without_boundingBox.txt)
2. Right-click on the created workshpace and select 'Show Instrument' 

then Mantid pops up the window

[image]

which shows a ga ga bounding box has been created for the hexahedron shape. T0 fix this permanently, i.e. avoid having a user to worry about bounding box for this shape all-together fix this issue #9034 . 

However while that is being fixed in this issue fix a bug when a bounding box is specified for hexahedron shape. To reproduce:

1. Run LoadEmptyInstrument with the IDF: [templateSANS_Mantid.instr_with_boundingBox.txt](https://github.com/mantidproject/mantid/files/29823/templateSANS_Mantid.instr_with_boundingBox.txt) 
2. Right-click on the created workshpace and select 'Show Instrument' 

then observed that the default instrument view is awful, and you need to zoom a lot to get to see any of the components defined in the IDF including the hexahedron shape. 

For some reason bounding box in 
```
<type name="MonNDtype-0-pix-4" is="detector">
	<hexahedron id="hexapix-4">
		<left-back-bottom-point  x="-0.0166667" y="-0.0336398" z="0"  />
		<left-front-bottom-point x="-0.0166667" y="0.0349515" z="0"  />
		<right-front-bottom-point x="0.0166667" y="0.0349515" z="0"  />
		<right-back-bottom-point  x="0.0166667" y="-0.0362632" z="0"  />
		<left-back-top-point  x="-0.0166667" y="-0.0336398" z="-0.001"  />
		<left-front-top-point  x="-0.0166667" y="0.0349515" z="-0.001"  />
		<right-front-top-point  x="0.0166667" y="0.0349515" z="-0.001"  />
		<right-back-top-point   x="0.0166667" y="-0.0362632" z="-0.001"  />
	</hexahedron>
<bounding-box>
    <x-min val="-0.02"/>
    <x-max val="0.02"/>
    <y-min val="-0.04"/>
    <y-max val="0.04"/>
    <z-min val="-0.002"/>
    <z-max val="0.002"/>
  </bounding-box>
</type>
```
 
is not used properly

MikeHart85, 2015-11-10:
Investigation showed that there are several issues at work here.

1. The reason that the ymin value for the hexahedron bounding box is off the charts is because the bottom surface of the hexahedron is not axis-aligned. All other sides match up with an axis-aligned plane, which is why their values are reasonable.  This is an artifact of the way that automatic bounding boxes are calculated. Hopefully this will be addressed in #9034. In the mean time, bounding boxes will have to be specified for any shape that has one or more surfaces that are not axis-aligned.

2. The reason the instrument still shows up poorly, even when a bounding box is given for the hexahedron, is that it isn't the only shape that wasn't axis-aligned. All of the line source cylinders have their axis defined with `<axis x="-0.00055556" y="-0.00055556" z="0.33333" />` tags or similar, making them slightly askew to the Z-axis. This causes an unreasonable bounding box to be created, for essentially the same reason as the hexahedron: the current system cannot handle shapes that aren't axis-aligned. A possible solution would to add bounding boxes to the cylinders. Another option would be to make them axis-aligned (and perhaps rotate them later, using the `rot` and `axis-[x,y,z]` properties of the `<location>` tag).

3. Finally, if you solve both of the above issues, the instrument still won't show up quite right. It will zoom in correctly, but the view will be moved sideways so not all shapes are immediately visible. This is due to a separate issue with the `axis-view` property of the `<default-view>` tag. I've tried it with a few instruments, and It seems that the only view that works correctly in all cases is `Z+`. All others (including `Z-` which this file uses) tend to show up with exactly the right zoom-level to show the entire instrument, but translated sideways so that parts of the instrument are out of view.

For now, I have updated the documentation page to better describe bounding box behaviour, and under what circumstances a custom one needs to be defined. I'll investigate a bit more to see what could be done to improve the current behaviour.

MikeHart85, 2015-11-12:
I've extended the bounding box calculation algorithm to be able to automatically construct good bounding boxes for off-axis and complex shapes.

As far as I can tell, it should be able to handle all possible shapes now, eliminating the need to ever specify a bounding box manually. If it does run into a shape it cannot construct a good bounding box for, it will behave as it did before: create a very large bounding box so that the instrument appears far away by default.

Although the calculations are fairly simple, we will need to keep an eye on performance. It may still be necessary to manually specify bounding boxes for shapes consisting of very many vertices. It might be a good idea to remove all bounding boxes from some of the larger and more complex instruments and run some performance tests to see how they compare.

MikeHart85, 2015-11-13:
Pull request to fix bounding boxes now open.

I've created a separate issue for Axis-View only working properly for Z+: #14402

PR #14424 (2015-11-17): Automatic Bounding Boxes for Non-Axis-Aligned Shapes
Fixes #14349 and #9034.

Release notes not yet updated.

This change expands Object::getBoundingBox() to support automatically calculating Axis-Aligned Bounding Boxes (AABB) for shapes that are not themselves axis-aligned. 

Previously, getBoundingBox would produce incorrect bounding boxes if a shape contained any surfaces that were not axis-aligned and required the user to provide explicit, manually calculated bounding boxes for all such shapes in the Instrument Definition File. This could be quite a hassle, especially since it was not alwa [... cut]
closed 2015-11-17 by mantid-roman