mantidproject/mantid #14461: Surface Plots from Group Workspaces

OwenArnold, 2015-11-17:
To create a surface plot around a group workspace we can do something like this (below) in the mantidplot script window.

However, it would be much more convenient if this functionality was built into MantidPlot. There should be a right-click on a group workspace that allows you to do this. The 3D qti plotting tools should be used. 

This request came from Pascal Manuel.

```python
from mpl_toolkits.mplot3d import Axes3D
from matplotlib import cm
from matplotlib.ticker import LinearLocator, FormatStrFormatter
import matplotlib.pyplot as plt
import numpy as np

import matplotlib as mpl
print mpl.__file__
    
def create_sample_group(n_items):
    to_group = list()
    for i in range(n_items):
        name = 'ws' + str(i)
        to_group.append(name)
        CreateWorkspace(DataX=range(200, 210, 1), DataY=[i+1]*10, NSpec=1, OutputWorkspace=name)
    group = GroupWorkspaces(InputWorkspaces=','.join(to_group))
    return group

ws = create_sample_group(n_items=10)
    
def make_graph(group_ws):
    
    sample = group_ws[0]
    X = sample.readX(0)
    Y = np.arange(0, group_ws.size(), 1)
    Z = (sample.readX(0), group_ws.size())
    X, Y = np.meshgrid(X, Y)
    
    z_arrays = list()
    for ws in group_ws:
        z_arrays.append(ws.readY(0))
    Z = np.array(z_arrays)
    print Z.shape

    fig = plt.figure()
    
    ax = fig.gca(projection='3d')
   
    
    surf = ax.plot_surface(X, Y, Z, rstride=1, cstride=1, cmap=cm.coolwarm,
                           linewidth=0, antialiased=False)
                           
    #ax.set_zlim(-1.01, 1.01)

    ax.zaxis.set_major_locator(LinearLocator(10))
    ax.zaxis.set_major_formatter(FormatStrFormatter('%.02f'))

    fig.colorbar(surf, shrink=0.5, aspect=5)

    plt.show()
    
to_plot = create_sample_group(10)   
threadsafe_call(make_graph, to_plot)
```

closed 2015-12-09 by KarlPalmen

OwenArnold, 2015-12-09:
@tom-perkins Has this been mentioned in the release notes for 3.6. Please include some screenshots.

tom-perkins, 2015-12-09:
The [release notes](http://www.mantidproject.org/ReleaseNotes_3_6_UI_Changes#Plotting_improvements) have now been updated with a screenshot.

PR #14730 (2015-12-09): Create a surface plot from a WorkspaceGroup
Resolves #14461 

Adds a right-click option to create a surface plot from a group workspace.

 - [x] Updated [release notes] (http://www.mantidproject.org/ReleaseNotes_3_6_UI_Changes#Plotting_improvements)

**To test:**
- Run the following to create a workspace group:
``` python
def create_sample_group(n_items):
    to_group = list()
    for i in range(n_items):
        name = 'ws' + str(i)
        to_group.append(name)
        CreateWorkspace(DataX=range(200, 210, 1), DataY=[i+1]*10, NSpec=1, OutputWorkspace=name)
    group = GroupWorkspaces(InputWo [... cut]