mantidproject/mantid #14463: Crash with FilterEvents Dialog

LamarMoore, 2015-11-17:
When running mantidplot in debug mode using Visual Studio 2015, loading the Interfaces->Utility->FilterEvents dialog from the main menu results in an Access Violation when exiting the software (even if FilterEvents has been closed):

[image]

NickDraper, 2015-11-17:
Note: This only occurs when the debugger is attached.  Code compiled in debug or release and run separately does not show this issue.

NickDraper, 2015-11-17:
Trying out the environment variable _NO_DEBUG_HEAP=1 does not affect this.
http://ofekshilon.com/2014/09/20/accelerating-debug-runs-part-1-_no_debug_heap-2/

NickDraper, 2015-11-17:
As this only happens on shutdown, and then only when debugging this has been given a low priority.

martyngigg, 2019-07-15:
This appears not to be an issue any longer.

closed 2019-07-15 by martyngigg