mantidproject/mantid #14467: GetDetOffsetsMultiPeaks does not finish on windows

FedeMPouzols, 2015-11-17:
Run the doc test of `GetDetOffsetsMultiPeaks `. This line never ends:
`msk = GetDetOffsetsMultiPeaks(ws,DReference=[5,15], GroupingFileName=calFilePath)` - the algorithm reaches 99% but does not finish.

- This happens in both Release and Debug mode in recent master code (today and last night too).
- In a build from last week it doesn't happen.
- I cannot reproduce this on debian.

The obvious suspect is the parallel nested loops of `GetDetOffsetsMultiPeaks::calculateDetectorsOffsets()`.

FedeMPouzols, 2015-11-17:
While testing something else, this doc test passed on win7-debug. Algorithm execution took 39.55. I wasn't looking at it but that seems a normal execution time in debug. On a second attempt it got stuck as usual, at 98%.

FedeMPouzols, 2015-11-18:
It is getting stuck on the Poco logger lock for thread-safe logging:

```
 	[External Code]	
>	MantidKernel.dll!Mantid::Kernel::ThreadSafeLogStreamBuf::writeToDevice(char c) Line 43	C++
 	PocoFoundation64d.dll!000007fed78daef7()	Unknown
 	[External Code]	
 	MantidCurveFitting.dll!Mantid::CurveFitting::CostFunctions::CostFuncLeastSquares::calActiveCovarianceMatrix(Mantid::CurveFitting::GSLMatrix & covar, double epsrel) Line 417	C++
 	MantidCurveFitting.dll!Mantid::CurveFitting::CostFunctions::CostFuncFitting::calCovarianceMatrix(Mantid::CurveFitting::GSLMatrix & covar, double epsrel) Line 126	C++
 	MantidCurveFitting.dll!Mantid::CurveFitting::Algorithms::Fit::execConcrete() Line 238	C++
 	MantidCurveFitting.dll!Mantid::CurveFitting::IFittingAlgorithm::exec() Line 284	C++
 	MantidAPI.dll!Mantid::API::Algorithm::execute() Line 587	C++
 	MantidAPI.dll!Mantid::API::Algorithm::executeAsChildAlg() Line 691	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FitOneSinglePeak::fitFunctionSD(boost::shared_ptr<Mantid::API::IFunction> fitfunc, boost::shared_ptr<Mantid::API::MatrixWorkspace> dataws, unsigned __int64 wsindex, double xmin, double xmax, bool calmode) Line 693	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FitOneSinglePeak::fitCompositeFunction(boost::shared_ptr<Mantid::API::IPeakFunction> peakfunc, boost::shared_ptr<Mantid::API::IBackgroundFunction> bkgdfunc, boost::shared_ptr<Mantid::API::MatrixWorkspace> dataws, unsigned __int64 wsindex, double startx, double endx) Line 840	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FitOneSinglePeak::highBkgdFit() Line 570	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FindPeaks::callFitPeak(const boost::shared_ptr<Mantid::API::MatrixWorkspace> & dataws, int wsindex, const boost::shared_ptr<Mantid::API::IPeakFunction> peakfunction, const boost::shared_ptr<Mantid::API::IBackgroundFunction> backgroundfunction, const std::vector<double,std::allocator<double> > & vec_fitwindow, const std::vector<double,std::allocator<double> > & vec_peakrange, int minGuessFWHM, int maxGuessFWHM, int guessedFWHMStep, int estBackResult) Line 1576	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FindPeaks::fitSinglePeak(const boost::shared_ptr<Mantid::API::MatrixWorkspace> & input, const int spectrum, const int i_min, const int i_max, const int i_centre) Line 1034	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FindPeaks::fitPeakGivenFWHM(const boost::shared_ptr<Mantid::API::MatrixWorkspace> & input, const int spectrum, const double center_guess, const int fitWidth, const bool hasleftpeak, const double leftpeakcentre, const bool hasrightpeak, const double rightpeakcentre) Line 885	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FindPeaks::findPeaksGivenStartingPoints(const std::vector<double,std::allocator<double> > & peakcentres, const std::vector<double,std::allocator<double> > & fitwindows) Line 416	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FindPeaks::exec() Line 198	C++
 	MantidAPI.dll!Mantid::API::Algorithm::execute() Line 587	C++
 	MantidAPI.dll!Mantid::API::Algorithm::executeAsChildAlg() Line 691	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::GetDetOffsetsMultiPeaks::fitSpectra(const __int64 wi, boost::shared_ptr<Mantid::API::MatrixWorkspace> inputW, const std::vector<double,std::allocator<double> > & peakPositions, const std::vector<double,std::allocator<double> > & fitWindows, unsigned __int64 & nparams, double & minD, double & maxD, std::vector<double,std::allocator<double> > & peakPosToFit, std::vector<double,std::allocator<double> > & peakPosFitted, std::vector<double,std::allocator<double> > & chisq, std::vector<double,std::allocator<double> > & peakHeights, int & i_highestpeak, double & resolution, double & dev_resolution) Line 923	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::GetDetOffsetsMultiPeaks::calculatePeakOffset(const int wi, std::vector<double,std::allocator<double> > & vec_peakPosFitted, std::vector<double,std::allocator<double> > & vec_peakPosRef) Line 621	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::GetDetOffsetsMultiPeaks::calculateDetectorsOffsets$omp$1() Line 491	C++
 	[External Code]	
```

When `CostFuncLeastSquares::calActiveCovarianceMatrix` does 
```
g_log.debug() << std::setw(10);
g_log.debug() << m_hessian.get(i, j) << "  ";
```

FedeMPouzols, 2015-11-19:
I'm puzzled by this possible dead-lock. This starts to look not like a dead-lock but a race condition or bug in the MSVC std streams.

The critical section in the main loop of `GetDetOffsetsMultiPeaks::calculateDetectorsOffsets()` is not to blame. You can comment it out, effectively leaving just the call to `calculatePeakOffset` in the parallel loop and you'll still get the maybe-dead-lock.

Now if in `CostFuncLeastSquares::calActiveCovarianceMatrix()` you take out the *two* lines that do `g_log.debug() << std::setw(10)` the issue is gone. It looks as if the operator called by those lines was causing an issue with the `Mantid::Kernel::ThreadSafeLogStreamBuf` mutex. All I can get with the MSVC debugger is that the operator is defined in `iomanip` as:
```
template<class _Elem,
	class _Traits,
	class _Arg> inline
	basic_ostream<_Elem, _Traits>& operator<<(
		basic_ostream<_Elem, _Traits>& _Ostr, const _Smanip<_Arg>& _Manip)
	{	// insert by calling function with output stream and argument
	(*_Manip._Pfun)(_Ostr, _Manip._Manarg);
	return (_Ostr);
	}
```

When I replace the `setw` with `setprecision` the issue also disappears. `setw` is not sticky, as opposed to other manipulators like `setprecision`, so

FedeMPouzols, 2015-11-19:
After managing to get more info from the debugger, this definitely looks like a race condition with the width parameter of the std stream.

In `<xlocnum>`, in code that outputs a number (in our case the elements of the hessian and covariance matrices), inside `_OutIt __CLRCALL_OR_CDECL _Fput(_OutIt _Dest,
		ios_base& _Iosbase, _Elem _Fill, const char *_Buf,
			size_t _Count) const` this code:
```
		size_t _Fillcount = _Iosbase.width() <= 0
			|| (size_t)_Iosbase.width() <= _Count
				? 0 : (size_t)_Iosbase.width() - _Count;
```
Can produce an extremely large number into `_Fillcount` which will then lead to an extremely long loop in here: 
```
	_OutIt __CLRCALL_OR_CDECL _Rep(_OutIt _Dest,
		_Elem _Ch, size_t _Count) const
		{	// put _Count * _Ch to _Dest
		for (; 0 < _Count; --_Count, ++_Dest)
			*_Dest = _Ch;
		return (_Dest);
		}
```
This `_Rep` method is being called with `_Fillcount` as the number of characters to write (`_Rep(_Dest, _Fill, _Fillcount)`)
In debug mode this is so slow that it can spend hours writing padding characters one at a time to the output . `_Fillcount` is being assigned for example `0-9` casted to a `size_t`.

FedeMPouzols, 2015-11-19:
This is an example trace where it is running the for loop with `_Count == 18446744073693097024`:

```
>	msvcp140d.dll!std::num_put<char,std::ostreambuf_iterator<char,std::char_traits<char> > >::_Rep(std::ostreambuf_iterator<char,std::char_traits<char> > _Dest, char _Ch, unsigned __int64 _Count) Line 1550	C++
 	msvcp140d.dll!std::num_put<char,std::ostreambuf_iterator<char,std::char_traits<char> > >::_Fput(std::ostreambuf_iterator<char,std::char_traits<char> > _Dest, std::ios_base & _Iosbase, char _Fill, const char * _Buf, unsigned __int64 _Count) Line 1447	C++
 	msvcp140d.dll!std::num_put<char,std::ostreambuf_iterator<char,std::char_traits<char> > >::do_put(std::ostreambuf_iterator<char,std::char_traits<char> > _Dest, std::ios_base & _Iosbase, char _Fill, double _Val) Line 1309	C++
 	msvcp140d.dll!std::num_put<char,std::ostreambuf_iterator<char,std::char_traits<char> > >::put(std::ostreambuf_iterator<char,std::char_traits<char> > _Dest, std::ios_base & _Iosbase, char _Fill, double _Val) Line 1200	C++
 	msvcp140d.dll!std::basic_ostream<char,std::char_traits<char> >::operator<<(double _Val) Line 448	C++
 	MantidCurveFitting.dll!Mantid::CurveFitting::CostFunctions::CostFuncLeastSquares::calActiveCovarianceMatrix(Mantid::CurveFitting::GSLMatrix & covar, double epsrel) Line 449	C++
 	MantidCurveFitting.dll!Mantid::CurveFitting::CostFunctions::CostFuncFitting::calCovarianceMatrix(Mantid::CurveFitting::GSLMatrix & covar, double epsrel) Line 126	C++
 	MantidCurveFitting.dll!Mantid::CurveFitting::Algorithms::Fit::execConcrete() Line 238	C++
 	MantidCurveFitting.dll!Mantid::CurveFitting::IFittingAlgorithm::exec() Line 284	C++
 	MantidAPI.dll!Mantid::API::Algorithm::execute() Line 587	C++
 	MantidAPI.dll!Mantid::API::Algorithm::executeAsChildAlg() Line 691	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FitOneSinglePeak::fitFunctionSD(boost::shared_ptr<Mantid::API::IFunction> fitfunc, boost::shared_ptr<Mantid::API::MatrixWorkspace> dataws, unsigned __int64 wsindex, double xmin, double xmax, bool calmode) Line 693	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FitOneSinglePeak::fitPeakFunction(boost::shared_ptr<Mantid::API::IPeakFunction> peakfunc, boost::shared_ptr<Mantid::API::MatrixWorkspace> dataws, unsigned __int64 wsindex, double startx, double endx) Line 472	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FitOneSinglePeak::highBkgdFit() Line 556	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FindPeaks::callFitPeak(const boost::shared_ptr<Mantid::API::MatrixWorkspace> & dataws, int wsindex, const boost::shared_ptr<Mantid::API::IPeakFunction> peakfunction, const boost::shared_ptr<Mantid::API::IBackgroundFunction> backgroundfunction, const std::vector<double,std::allocator<double> > & vec_fitwindow, const std::vector<double,std::allocator<double> > & vec_peakrange, int minGuessFWHM, int maxGuessFWHM, int guessedFWHMStep, int estBackResult) Line 1576	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FindPeaks::fitSinglePeak(const boost::shared_ptr<Mantid::API::MatrixWorkspace> & input, const int spectrum, const int i_min, const int i_max, const int i_centre) Line 1034	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FindPeaks::fitPeakGivenFWHM(const boost::shared_ptr<Mantid::API::MatrixWorkspace> & input, const int spectrum, const double center_guess, const int fitWidth, const bool hasleftpeak, const double leftpeakcentre, const bool hasrightpeak, const double rightpeakcentre) Line 885	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FindPeaks::findPeaksGivenStartingPoints(const std::vector<double,std::allocator<double> > & peakcentres, const std::vector<double,std::allocator<double> > & fitwindows) Line 416	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::FindPeaks::exec() Line 198	C++
 	MantidAPI.dll!Mantid::API::Algorithm::execute() Line 587	C++
 	MantidAPI.dll!Mantid::API::Algorithm::executeAsChildAlg() Line 691	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::GetDetOffsetsMultiPeaks::fitSpectra(const __int64 wi, boost::shared_ptr<Mantid::API::MatrixWorkspace> inputW, const std::vector<double,std::allocator<double> > & peakPositions, const std::vector<double,std::allocator<double> > & fitWindows, unsigned __int64 & nparams, double & minD, double & maxD, std::vector<double,std::allocator<double> > & peakPosToFit, std::vector<double,std::allocator<double> > & peakPosFitted, std::vector<double,std::allocator<double> > & chisq, std::vector<double,std::allocator<double> > & peakHeights, int & i_highestpeak, double & resolution, double & dev_resolution) Line 929	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::GetDetOffsetsMultiPeaks::calculatePeakOffset(const int wi, std::vector<double,std::allocator<double> > & vec_peakPosFitted, std::vector<double,std::allocator<double> > & vec_peakPosRef) Line 623	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::GetDetOffsetsMultiPeaks::calculateDetectorsOffsets$omp$1() Line 491	C++
 	[External Code]	
 	MantidAlgorithms.dll!Mantid::Algorithms::GetDetOffsetsMultiPeaks::calculateDetectorsOffsets() Line 553	C++
 	MantidAlgorithms.dll!Mantid::Algorithms::GetDetOffsetsMultiPeaks::exec() Line 263	C++
 	MantidAPI.dll!Mantid::API::Algorithm::execute() Line 587	C++
 	MantidAPI.dll!Mantid::API::AlgorithmProxy::execute() Line 71	C++
 	_api.pyd!`anonymous namespace'::executeProxy(boost::python::api::object & self) Line 258	C++
[... cut ...]
 	python27.dll!000000001e119782()	Unknown
 	python27.dll!000000001e119819()	Unknown
 	MantidPlot.exe!PythonScript::executeCompiledCode(_object * compiledCode) Line 765	C++
 	MantidPlot.exe!PythonScript::executeString() Line 693	C++
 	MantidPlot.exe!PythonScript::executeImpl() Line 648	C++
 	MantidPlot.exe!Script::execute(const ScriptCode & code) Line 151	C++
 	MantidPlot.exe!Script::ScriptTask::run() Line 62	C++
 	QtCored4.dll!000000006652653d()	Unknown
 	QtCored4.dll!000000006654d07f()	Unknown
 	[External Code]	
```

FedeMPouzols, 2015-11-19:
@martyngigg : I've been chasing this and came to the conclusion that the algorithm doesn't finish in debug mode because one or more threads get trapped in an extremely long loop that writes a ridiculously large padding (which runs very slowly in debug mode).
A couple of `g_log.debug() << setw(10);` are to blame. There is a race condition with the width parameter of the base ostream. Have you seen similar issues before?

The issue goes away if we remove the `<< setw(10)`, or we put an ad-hoc ScopedLock guarding the loops that log the hessian and covariance matrices in `CostFuncLeastSquares::calActiveCovarianceMatrix()`. But I'd worry that this is a more general issue that can become a recurrent source of headaches if we use manipulators with the Mantid loggers. Was `ThreadSafeLogStreamBuf` meant to be used with manipulators?

martyngigg, 2015-11-19:
In theory there should be nothing preventing their use but obviously something is not quite right.

It looks to me as if in [CostFuncLeastSquares::calActiveCovarianceMatrix()](https://github.com/mantidproject/mantid/blob/f821234b70f872355f8be2a3735e17d00f31ff40/Framework/CurveFitting/src/CostFunctions/CostFuncLeastSquares.cpp#L410) the logging should only happen if the level is set to debug? Is that happening?

martyngigg, 2015-11-19:
A different way of doing the logging might be to use a `std::ostringstream` to build up message and then do a single dispatch to the logger using the other `g_log` debug method that accepts a string:
```c++
std::ostringstream os;
// Build up message...
g_log.debug(os.str());
```

FedeMPouzols, 2015-11-19:
The messages are being written character by character through [`ThreadSafeLogStreamBuf::writeToDevice`](https://github.com/mantidproject/mantid/blob/master/Framework/Kernel/src/ThreadSafeLogStream.cpp#L35). They are accumulated in `ThreadSafeLogStreamBuf::m_messages` even if the log level is less detailed than debug. Then it seems that the logger will output or discard them depending on their priority value, but only when a newline arrives. So race conditions can happen regardless of the current log level.

Oh and yes, building up the message with a std::ostringstream looks much neater.

PR #14539 (2015-11-23): Change debug logs in Fit / CostFuncLeastSquares (to fix doc test of GetDetSetOffset on win7-debug)
Fixes #14467.

The doc test of GetDetOffsetsMultiPeaks was getting stuck on windows (debug build) >95% of times because of a data race condition in stream manipulators when writing debug messages concurrently.

**To test:**
- The doctest GetDetOffsetsMultiPeaks should work (finish) in a windows debug build.
- Check that if you set 'debug' logging level you still see the very verbose output from the underlying Fit runs, including the Hessian  and covariance matrices.

Note that the output messages [... cut]
closed 2015-11-23 by MattKing06