mantidproject/mantid #14490: Add Poisson cost function

mantid-roman, 2015-11-18:
The correct cost function formula for Poisson statistics should be (for bin contents=N, fitted mean count rate=y)
y<0: forbidden (infinite cost)
y>=0, N=0: cost function=2*y
y>=0, N>0: cost function=2*( (y-N) + N*(log(N)-log(y)) )

Anders-Markvardsen, 2016-02-15:
Just looking at this ticket with Joe.

Fixing this might do what Joe needs for fitting a background.

As a comment why is there not a factor of 2 to much, at least when comparing to -log(lambda^N*exp(-lambda)/lambda!) using the stirling approximation.

Please inform Joe when this is fixed

mantid-roman, 2016-02-15:
At the moment it works for functions of the form y = exp( f(x) ) or any other strictly positive function.

PR #26755 (2019-09-17): Poisson Cost Function
**Description of work.**
This finishes Roman's work on Poisson cost functions (started in #25686)

@alicerussell1 could you add release notes if required, as you are more familiar with the potential benefits of this technique.

Changes since the original PR:
- Squashes the original commit series and resolve conflicts with master
- Added unit tests for the methods
- Moved protected methods into public following the conventions of the base classes
- Fixed a bug where using any parameters would result in an OOB
- Fixed a bug where the determinant was always calculated instea [... cut]
closed 2019-09-17 by peterfpeterson