mantidproject/mantid #14514: RecordPythonScript doc test fails on win7-debug

FedeMPouzols, 2015-11-19:
Something goes wrong with the threads. On other platforms a thread is also terminated badly, but the test works. On debian I can see the "Execution terminated by user" error and the `KeyboardInterrupt` exception but that doesn't prevent the test from succeeding.

Run it with `.\bin\Debug\MantidPlot.exe -xq docs\runsphinx_doctest.py -R RecordPythonScript` and you'll get:
```
running tests...

Document: algorithms/RecordPythonScript-v1
------------------------------------------
RecordPythonScript-[Notice] RecordPythonScript started
CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.17 s
econds
CreateFlatEventWorkspace-[Notice] CreateFlatEventWorkspace started
CreateFlatEventWorkspace-[Notice] CreateFlatEventWorkspace successful, Duration
0.73 seconds
RebinToWorkspace-[Notice] RebinToWorkspace started
RebinToWorkspace-[Notice] RebinToWorkspace successful, Duration 0.01 seconds
RecordPythonScript-[Error] RecordPythonScript: Execution terminated by user.
Exception in thread Thread-6:
Traceback (most recent call last):
  File "D:\test\mantid-repos\mantid\external\src\ThirdParty\lib\python2.7\lib\th
reading.py", line 810, in __bootstrap_inner
    self.run()
  File "D:\test\mantid-repos\mantid\external\src\ThirdParty\lib\python2.7\lib\th
reading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "<doctest RecordPythonScript[0]>", line 13, in startRecording
    RecordPythonScript(outputFile)
  File "D:/test/build-mantid-release/bin/Debug\mantid\simpleapi.py", line 822, i
n algorithm_wrapper
    algm.execute()
KeyboardInterrupt

**********************************************************************
File "algorithms\RecordPythonScript-v1.rst", line 45, in RecordPythonScript
Failed example:
    from threading import Thread
    import os, time

    #find a suitable directory to save the file
    fileDir = config["defaultsave.directory"]
    if not os.path.isdir(fileDir):
        #use the users home directory if default save is not set
        fileDir = os.path.expanduser('~')
    outputFile = os.path.join(fileDir,"MyRecording.py")

    def startRecording():
        try:
            RecordPythonScript(outputFile)
        except RuntimeError:
            pass
    thread = Thread(target = startRecording)
    thread.start()

    # a short pause to allow the thread to start
    time.sleep(0.1)

    ws = CreateSampleWorkspace("Event","Multiple Peaks")
    wsOut=CreateFlatEventWorkspace(ws,RangeStart=15000,RangeEnd=18000)
    wsOut=RebinToWorkspace(wsOut,ws,PreserveEvents=True)

    # This will cancel the rocording algorithm
    # you can do the same in the GUI
    # by clicking on the details button on the bottom right
    AlgorithmManager.newestInstanceOf("RecordPythonScript").cancel()
    thread.join()

    #Load and print the resulting file
    print "The result file has the following python recorded"
    with open(outputFile, "r") as file:
        print file.read().rstrip()

    #cleanup
    os.remove(outputFile)
Expected:
    The result file has the following python recorded
    CreateSampleWorkspace(OutputWorkspace='ws',WorkspaceType='Event',Function='M
ultiple Peaks',UserDefinedFunction='',NumBanks='2',BankPixelWidth='10',NumEvents
='1000',Random='0',XUnit='TOF',XMin='0',XMax='20000',BinWidth='200',PixelSpacing
='0.0080000000000000002',BankDistanceFromSample='5',SourceDistanceFromSample='10
')
    CreateFlatEventWorkspace(InputWorkspace='ws',RangeStart='15000',RangeEnd='18
000',OutputWorkspace='wsOut')
    RebinToWorkspace(WorkspaceToRebin='wsOut',WorkspaceToMatch='ws',OutputWorksp
ace='wsOut',PreserveEvents='1')
Got:
    The result file has the following python recorded
    CreateFlatEventWorkspace(InputWorkspace='ws',RangeStart='15000',RangeEnd='18
000',OutputWorkspace='wsOut')
    RebinToWorkspace(WorkspaceToRebin='wsOut',WorkspaceToMatch='ws',OutputWorksp
ace='wsOut',PreserveEvents='1')
**********************************************************************
1 items had failures:
   1 of   1 in RecordPythonScript
1 tests in 1 items.
0 passed and 1 failed.
***Test Failed*** 1 failures.
1 items passed all tests:
   1 tests in RecordPythonScript (cleanup code)
1 tests in 1 items.
1 passed and 0 failed.
Test passed.

Doctest summary
===============
    1 test
    1 failure in tests
    0 failures in setup code
    0 failures in cleanup code
build finished with problems.
```

MattKing06, 2015-11-19:
This issue also appears for StartLiveData-v1 on Ubuntu docs build:

Document: algorithms/StartLiveData-v1
12:34:25 -------------------------------------
12:34:25 FakeISISHistoDAE-[Fatal] FakeISISHistoDAE: Execution terminated by user.
12:34:25 Exception in thread Thread-8:
12:34:25 Traceback (most recent call last):
12:34:25   File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
12:34:25     self.run()
12:34:25   File "/usr/lib/python2.7/threading.py", line 763, in run
12:34:25     self.__target(*self.__args, **self.__kwargs)
12:34:25   File "<doctest exStartLiveDataEvent[0]>", line 9, in startFakeDAE
12:34:25     FakeISISEventDAE(NPeriods=1,NSpectra=100,Rate=20,NEvents=1000)
12:34:25   File "/home/builder/Jenkins/workspace/pull_requests-ubuntu/build/bin/mantid/simpleapi.py", line 822, in algorithm_wrapper
12:34:25     algm.execute()
12:34:25 KeyboardInterrupt
12:34:25 
12:34:25 Exception in thread Thread-9:
12:34:25 Traceback (most recent call last):
12:34:25   File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
12:34:25     self.run()
12:34:25   File "/usr/lib/python2.7/threading.py", line 763, in run
12:34:25     self.__target(*self.__args, **self.__kwargs)
12:34:25   File "<doctest exStartLiveDataHisto[0]>", line 8, in startFakeDAE
12:34:25     FakeISISHistoDAE(NPeriods=5,NSpectra=10,NBins=100)
12:34:25   File "/home/builder/Jenkins/workspace/pull_requests-ubuntu/build/bin/mantid/simpleapi.py", line 822, in algorithm_wrapper
12:34:25     algm.execute()
12:34:25 KeyboardInterrupt
12:34:25 
12:34:25 2 items passed all tests:
12:34:25    1 tests in exStartLiveDataEvent
12:34:25    1 tests in exStartLiveDataHisto
12:34:25 2 tests in 2 items.
12:34:25 2 passed and 0 failed.
12:34:25 Test passed.
12:34:25 2 items passed all tests:
12:34:25    1 tests in exStartLiveDataEvent (cleanup code)
12:34:25    1 tests in exStartLiveDataHisto (cleanup code)
12:34:25 2 tests in 2 items.
12:34:25 2 passed and 0 failed.
12:34:25 Test passed.

FedeMPouzols, 2016-02-09:
This is still happening.
To reproduce just run it in debug mode:
`bin\Debug\MantidPlot.exe -xq docs\runsphinx_doctest.py`:

```
Document: algorithms/RecordPythonScript-v1
------------------------------------------
RecordPythonScript-[Notice] RecordPythonScript started
CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.20 s
econds
CreateFlatEventWorkspace-[Notice] CreateFlatEventWorkspace started
CreateFlatEventWorkspace-[Notice] CreateFlatEventWorkspace successful, Duration
1.83 seconds
RebinToWorkspace-[Notice] RebinToWorkspace started
RebinToWorkspace-[Notice] RebinToWorkspace successful, Duration 0.01 seconds
RecordPythonScript-[Error] RecordPythonScript: Execution terminated by user.
Exception in thread Thread-6:
Traceback (most recent call last):
  File "D:\test\mantid-repos\mantid\external\src\ThirdParty\lib\python2.7\lib\th
reading.py", line 810, in __bootstrap_inner
    self.run()
  File "D:\test\mantid-repos\mantid\external\src\ThirdParty\lib\python2.7\lib\th
reading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "<doctest RecordPythonScript[0]>", line 13, in startRecording
    RecordPythonScript(outputFile)
  File "D:/test/build-mantid-debug/bin/Debug\mantid\simpleapi.py", line 833, in
algorithm_wrapper
    algm.execute()
KeyboardInterrupt

**********************************************************************
File "algorithms\RecordPythonScript-v1.rst", line 45, in RecordPythonScript
Failed example:
    from threading import Thread
    import os, time

    #find a suitable directory to save the file
    fileDir = config["defaultsave.directory"]
    if not os.path.isdir(fileDir):
        #use the users home directory if default save is not set
        fileDir = os.path.expanduser('~')
    outputFile = os.path.join(fileDir,"MyRecording.py")

    def startRecording():
        try:
            RecordPythonScript(outputFile)
        except RuntimeError:
            pass
    thread = Thread(target = startRecording)
    thread.start()

    # a short pause to allow the thread to start
    time.sleep(0.1)

    ws = CreateSampleWorkspace("Event","Multiple Peaks")
    wsOut=CreateFlatEventWorkspace(ws,RangeStart=15000,RangeEnd=18000)
    wsOut=RebinToWorkspace(wsOut,ws,PreserveEvents=True)

    # This will cancel the rocording algorithm
    # you can do the same in the GUI
    # by clicking on the details button on the bottom right
    AlgorithmManager.newestInstanceOf("RecordPythonScript").cancel()
    thread.join()

    #Load and print the resulting file
    print "The result file has the following python recorded"
    with open(outputFile, "r") as file:
        print file.read().rstrip()

    #cleanup
    os.remove(outputFile)
Expected:
    The result file has the following python recorded
    CreateSampleWorkspace(OutputWorkspace='ws',WorkspaceType='Event',Function='M
ultiple Peaks',UserDefinedFunction='',NumBanks='2',BankPixelWidth='10',NumEvents
='1000',Random='0',XUnit='TOF',XMin='0',XMax='20000',BinWidth='200',PixelSpacing
='0.0080000000000000002',BankDistanceFromSample='5',SourceDistanceFromSample='10
')
    CreateFlatEventWorkspace(InputWorkspace='ws',RangeStart='15000',RangeEnd='18
000',OutputWorkspace='wsOut')
    RebinToWorkspace(WorkspaceToRebin='wsOut',WorkspaceToMatch='ws',OutputWorksp
ace='wsOut',PreserveEvents='1')
Got:
    The result file has the following python recorded
    CreateFlatEventWorkspace(InputWorkspace='ws',RangeStart='15000',RangeEnd='18
000',OutputWorkspace='wsOut')
    RebinToWorkspace(WorkspaceToRebin='wsOut',WorkspaceToMatch='ws',OutputWorksp
ace='wsOut',PreserveEvents='1')
**********************************************************************
1 items had failures:
   1 of   1 in RecordPythonScript
1 tests in 1 items.
0 passed and 1 failed.
***Test Failed*** 1 failures.
```

FedeMPouzols, 2016-08-12:
This has been fixed since then. This test hasn't failed in the debug builds for a long time.

closed 2016-10-03 by NickDraper