mantidproject/mantid #14516: Refactoring of merging REAR and FRONT data in SANS

picatess, 2015-11-19:
The ISIS SANS reduction has the ability to stitch together the results of the FRONT reduction and the REAR reduction. 

The relevant locations in the code are [here](https://github.com/mantidproject/mantid/blob/master/scripts/SANS/ISISCommandInterface.py#L461:L548) and  [here](https://github.com/mantidproject/mantid/blob/master/scripts/SANS/ISISCommandInterface.py#L585:L670). 
Both code segments don't depend on the `ReductionSingleton`. This suggests that the code can be refactored out into a separate merge algorithm. 

This algorithm would also potentially contain the fitting routine which determines the correct scale and shift to stitch the two worksapces together. Note though that this functionality is also required elsewhere in the code and hence needs to stay exposed in some way.

The bare merge algorithm would need WS1 and WS2 as its input as well as the partial workspaces which are obtianed from Q1D, ie the count and normalisation workspaces.

The fitting bit requires the  two workspaces WS1 and WS2 and an initial value for the scale and the shift.

OwenArnold, 2015-11-23:
Parameters and naming:

ScaleFactor : double (optional)
ShiftFactor : double (optional)
Mode : {ScaleOnly, ShiftOnly, Both}
HABCounts : MatrixWorkspace
HABNorm : MatrixWorkspace
LABCounts : MatrixWorkspace
LABNorm : MatrixWorkspace
QMin : double (optional)
QMax : double (optional)

picatess, 2015-11-23:
__Current Fitting__

The fit we model that is used by the instrument scientists is `f(x) = a(x + b)`, where a(scale) and b(shift) can be free fit parameters. Note that the Mantid fit function is of the type `F(x) = A*x + B`, where `Ax` is of the TabulatedFuncion type and `B` is a FlatBackground. These two fit function types are just added.

There are four use-cases

1. `a` and `b` are both fixed:
In this case we don't require fitting and use the fixed values.

2. `a` is fixed and `b` is free:
In this case `B = a*b` for the free parameter of our fit function, where a is constant. Note that at the end of the fitting procedure we always perform `shift = B/a`. This means that `shift = b` which is the correct, fitted value

3. a is free and b is fixed:
In this case `A = a` and `B= a*b` as the free parameters(with `b` being a constant). Again, the operation `shift = B/a` produces the correct, fixed value. Note in the fit function we need to specify `a*b` explicitly, ie produce a interdependency between the two fit function types.

4. `a` is free and `b` is free:
This is the easiest case and we don' thave to specify anything in particular

Note that the two input workspaces need an overlap region in Q-space for the fitting to work.

picatess, 2015-11-23:
Mode should also allow for fixed scale and shifting
Mode : {None, ScaleOnly, ShiftOnly, Both}

OwenArnold, 2015-11-23:
@picatess. Thanks for your comments.

PR #14727 (2015-12-16): SANS Stitch1D Algorithm
Fixes #14516

**Tester**

Should check implementation and tests. Should try scratch-refactor to make ISIS sans command interface use this algorithm. If that works, then should give higher confidence.
closed 2015-12-16 by picatess